luxonis/depthai-ros #1: Branch develop_merge_config_manager no longer exists on depthai repo

job-1994, 2021-01-31:
Hi, I ran into a bunch of issues trying to get this to run:

1. Git clone of this repo and the depthai according to the instruction on this Readme come up with:

Cloning into 'depthai'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.
Please make sure you have the correct access rights
and the repository exists.

Problem averted by simply running `git clone https://github.com/luxonis/depthai_ros2.git`

2. Next problem was the depthai.depthai_helpers.arg_manager imports SharedArgs module, which does not exist (at least not in the develop branch of depthai, this may be why the specific branch is needed, but since i cant find it i'm not sure). 

This problem could be averted by removing the import, as it does not seem to be used anywhere.

3. This is where my breaking problem is. I get the following error when i run `ros2 run depthai_wrapper talker`

Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/job/ros2_ws/install/depthai_wrapper/lib/python3.6/site-packages/depthai_wrapper/publisher_member_function.py", line 152, in thread_function
    serializedEntry = self.decode_nn_json(nnet_packet, config=self.config)
  File "/home/job/depthai/depthai_helpers/mobilenet_ssd_handler.py", line 68, in decode_mobilenet_ssd_json
    detections = decode_mobilenet_ssd(nnet_packet, **kwargs)
  File "/home/job/depthai/depthai_helpers/mobilenet_ssd_handler.py", line 7, in decode_mobilenet_ssd
    NN_metadata = kwargs['NN_json']
KeyError: 'NN_json'

I have printed out the kwargs dict and there clearly isnt any NN_json key, however when I run the depthai_demo.py from the depthai repo it works just fine, and the dict shows the NN_json key filled in.

Question is, did one of my other changes break it, or is something else wrong?

PS: Full output of command is below. Running on Jetson Nano 4Gb, running ubuntu 18.04, ROS2 Eloquent

No calibration file. Using Calibration Defaults.
depthai.__version__ == 0.4.1.1+dev
{'config_overwrite': None, 'board': None, 'shaves': None, 'cmx_slices': None, 'NN_engines': None, 'model_compilation_target': 'auto', 'rgb_resolution': 1080, 'rgb_fps': 30.0, 'color_scale': 1.0, 'mono_resolution': 400, 'mono_fps': 30.0, 'disparity_confidence_threshold': 200, 'stereo_median_size': 7, 'stereo_lr_check': False, 'store_eeprom': False, 'clear_eeprom': False, 'override_eeprom': False, 'device_id': '', 'dev_debug': None, 'force_usb2': None, 'cnn_model': 'mobilenet-ssd', 'cnn_model2': '', 'cnn_camera': 'rgb', 'disable_depth': False, 'draw_bb_depth': False, 'full_fov_nn': False, 'sync_video_meta': False, 'sync_sequence_numbers': False, 'usb_chunk_KiB': 64, 'firmware': None, 'verbose': False, 'streams': ['metaout', 'previewout'], 'video': None, 'pointcloud': False, 'use_mesh': False, 'mirror_rectified': 'true', 'field_of_view': 71.86, 'rgb_field_of_view': 68.7938, 'baseline': 7.5, 'rgb_baseline': 3.75, 'swap_lr': True}
Compiled mode found: compiled for 14 shaves, 14 cmx_slices and 1 NN_engines 
XLink initialized.
Sending internal device firmware
Successfully connected to device.
Usb speed : Super/5000Mbps
Mx serial id : 14442C10C14790D000
Loading config file
Attempting to open stream config_d2h
watchdog started 
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
EEPROM data: valid (v5)
  Board name     : BW1098OBC
  Board rev      : R0M0E0
  HFOV L/R       : 71.86 deg
  HFOV RGB       : 68.7938 deg
  L-R   distance : 7.5 cm
  L-RGB distance : 3.75 cm
  L/R swapped    : yes
  L/R crop region: center
  Rectification Rotation R1 (left):
    0.999989,   -0.004569,   -0.001338,
    0.004571,    0.999989,    0.001381,
    0.001332,   -0.001387,    0.999998,
  Rectification Rotation R2 (right):
    0.999971,   -0.001382,   -0.007476,
    0.001371,    0.999998,   -0.001389,
    0.007478,    0.001379,    0.999971,
  Calibration intrinsic matrix M1 (left):
[... 103 lines cut ...]
E: [global] [    450516] [talker] addEvent:264	Condition failed: event->header.flags.bitField.ack != 1
E: [global] [    450516] [talker] addEventWithPerf:276	 addEvent(event) method call failed with an error: 3
E: [global] [    450516] [talker] XLinkReadData:156	Condition failed: (addEventWithPerf(&event, &opTime))
Device get data failed: 7
Closing stream previewout: ...
E: [global] [    450528] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -1) | event XLINK_CLOSE_STREAM_REQ

E: [xLink] [    450528] [Scheduler00Thr] sendEvents:1027	Event sending failed
Closing stream metaout: DONE.
Thread for metaout finished.
E: [global] [    450552] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -1) | event XLINK_CLOSE_STREAM_REQ

E: [xLink] [    450552] [Scheduler00Thr] sendEvents:1027	Event sending failed
Closing stream previewout: DONE.
Thread for previewout finished.
Stopping threads: DONE 0.049s.
Closing all observer streams: ...
Closing stream host_capture: ...
Closing stream host_capture: DONE.
Closing all observer streams: DONE.
Reseting device: 0.
E: [global] [    450588] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -1) | event XLINK_RESET_REQ

E: [xLink] [    450589] [Scheduler00Thr] sendEvents:1027	Event sending failed
Reseting: DONE.

Luxonis-Brandon, 2021-02-01:
I think @jonngai is likely best to help here.  Sorry about the trouble.

jonngai, 2021-02-01:
Hi @job-1994, sorry, it looks like the depthai repo has diverged a bit since this was written. It also looks like the commit I was referencing in the instructions got deleted. Could you give it a try with this depthai commit?

git checkout b6246bb16749c0a26ffa0ba78cd351f32e953d41

Thanks.

job-1994, 2021-02-01:
I get two issues with that, firstly the NN blob complains about not being found (path ../depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.blob). Found a similar named file but with an appended name (...blob.sh14cmx...), renamed it without the appendix. This seems to work, but then it complains about code in depathai.py not having certain features (depthai.__version__ does not exist). Im a bit worried that this will be a bit of a rabbit hole in terms of getting it to work. What is the (rough) time estimate of this new ros2 wrapper? I may try and get the ros wrapper to work with the latest version of depthai, with a little direction I reckon I could get it to work

jonngai, 2021-02-01:
@saching13 is working on the new ROS2 interface so he'd have a better estimate on timeframe for that. I'll install ROS2 back on my machine to take a look at what's going on with this branch too.

jonngai, 2021-02-02:
Btw, @saching13 , I just checked in some changes that should help the example run on the latest in develop (391bc4c1032ab94b7c31fece6bdf619bbdc3629b)

saching13, 2021-02-02:
Thanks. And sorry I din't see it before. 
Complete pipeline might take some time. but will have the basic features and networks soon.

jonngai, 2021-02-02:
Oh, ha, sorry, that second comment was actually meant for @job-1994: 

> Btw, @job-1994 , I just checked in some changes that should help the example run on the latest in develop (391bc4c1032ab94b7c31fece6bdf619bbdc3629b)

I tagged the wrong person.

closed 2021-03-19 by saching13