luxonis/depthai-ros #2: catkin_make error

ans-qureshi, 2021-04-02:
Hi!
I have run the command: `catkin_make_isolated --cmake-args -D depthai_DIR=$/home/srl/depthai-core/build/install/lib/cmake/depthai/` and I'm getting this error:

[image]

saching13, 2021-04-02:
Hi, 
Did steps from 1-6 work without any errors?

ans-qureshi, 2021-04-02:
@saching13 Yess, 1-15 worked without any errors. 
I am using Ubuntu 20.04
P.S. is the depthai_DR path that I gave correct?

themarpe, 2021-04-02:
@ans-qureshi 2 things that come to mind:
- does the folder contain the specified file? Was the install command run?
- you may also try pointing to build directory and just build (without installing)

ans-qureshi, 2021-04-02:
1. Folder contains specified file
2. Install command from point 7 was run
3. Tried pointing to build directory only, same issue :/

ans-qureshi, 2021-04-02:
@themarpe

saching13, 2021-04-02:
Your command should be 
`catkin_make_isolated --cmake-args -D depthai_DIR=/home/srl/depthai-core/build/install/lib/cmake/depthai/`

Not 
`catkin_make_isolated --cmake-args -D depthai_DIR=$/home/srl/depthai-core/build/install/lib/cmake/depthai/`

There is a `$` after depthai_DIR=. 
$ is used when representing the path using a variable.

ans-qureshi, 2021-04-02:
Sadly, no effect:

`srl@srl-HP-ProBook-440-G5:~/ros_ws$  catkin_make_isolated --cmake-args -D depthai_DIR=$/home/srl/depthai-core/build/install/lib/cmake/depthai/
Base path: /home/srl/ros_ws
Source space: /home/srl/ros_ws/src
Build space: /home/srl/ros_ws/build_isolated
Devel space: /home/srl/ros_ws/devel_isolated
Install space: /home/srl/ros_ws/install_isolated
Additional CMake Arguments: -D depthai_DIR=$/home/srl/depthai-core/build/install/lib/cmake/depthai/
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 4 packages in topological order:
~~  - vision_msgs
~~  - depthai_ros_msgs
~~  - depthai_bridge
~~  - depthai_examples
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
The packages or cmake arguments have changed, forcing cmake invocation

==> Processing catkin package: 'vision_msgs'
==> cmake /home/srl/ros_ws/src/vision_msgs -DCATKIN_DEVEL_PREFIX=/home/srl/ros_ws/devel_isolated/vision_msgs -DCMAKE_INSTALL_PREFIX=/home/srl/ros_ws/install_isolated -D depthai_DIR=$/home/srl/depthai-core/build/install/lib/cmake/depthai/ -G Unix Makefiles in '/home/srl/ros_ws/build_isolated/vision_msgs'
-- Using CATKIN_DEVEL_PREFIX: /home/srl/ros_ws/devel_isolated/vision_msgs
-- Using CMAKE_PREFIX_PATH: /home/srl/catkin_ws/devel;/home/srl/ros_ws/devel;/opt/ros/noetic
-- This workspace overlays: /home/srl/catkin_ws/devel;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.5", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/srl/ros_ws/build_isolated/vision_msgs/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.5") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- vision_msgs: 12 messages, 0 services
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    depthai_DIR

-- Build files have been written to: /home/srl/ros_ws/build_isolated/vision_msgs
==> make -j8 -l8 in '/home/srl/ros_ws/build_isolated/vision_msgs'
[  0%] Built target _vision_msgs_generate_messages_check_deps_ObjectHypothesisWithPose
[  0%] Built target _vision_msgs_generate_messages_check_deps_Detection2DArray
[  0%] Built target _vision_msgs_generate_messages_check_deps_Detection3D
[  0%] Built target _vision_msgs_generate_messages_check_deps_Detection3DArray
[  0%] Built target _vision_msgs_generate_messages_check_deps_Detection2D
[  0%] Built target _vision_msgs_generate_messages_check_deps_BoundingBox3D
[  0%] Built target _vision_msgs_generate_messages_check_deps_Classification3D
[  0%] Built target _vision_msgs_generate_messages_check_deps_VisionInfo
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _vision_msgs_generate_messages_check_deps_ObjectHypothesis
[  0%] Built target _vision_msgs_generate_messages_check_deps_Classification2D
[  0%] Built target _vision_msgs_generate_messages_check_deps_BoundingBox2D
[  0%] Built target _vision_msgs_generate_messages_check_deps_BoundingBox3DArray
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_eus
[... 94 lines cut ...]
CMake Error at CMakeLists.txt:82 (find_package):
  Could not find a package configuration file provided by "depthai" with any
  of the following names:

    depthaiConfig.cmake
    depthai-config.cmake

  Add the installation prefix of "depthai" to CMAKE_PREFIX_PATH or set
  "depthai_DIR" to a directory containing one of the above files.  If
  "depthai" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/srl/ros_ws/build_isolated/depthai_bridge/CMakeFiles/CMakeOutput.log".
See also "/home/srl/ros_ws/build_isolated/depthai_bridge/CMakeFiles/CMakeError.log".
<== Failed to process package 'depthai_bridge': 
  Command '['/home/srl/ros_ws/devel_isolated/depthai_ros_msgs/env.sh', 'cmake', '/home/srl/ros_ws/src/depthai-ros/depthai_bridge', '-DCATKIN_DEVEL_PREFIX=/home/srl/ros_ws/devel_isolated/depthai_bridge', '-DCMAKE_INSTALL_PREFIX=/home/srl/ros_ws/install_isolated', '-D', 'depthai_DIR=$/home/srl/depthai-core/build/install/lib/cmake/depthai/', '-G', 'Unix Makefiles']' returned non-zero exit status 1.

Reproduce this error by running:
==> cd /home/srl/ros_ws/build_isolated/depthai_bridge && /home/srl/ros_ws/devel_isolated/depthai_ros_msgs/env.sh cmake /home/srl/ros_ws/src/depthai-ros/depthai_bridge -DCATKIN_DEVEL_PREFIX=/home/srl/ros_ws/devel_isolated/depthai_bridge -DCMAKE_INSTALL_PREFIX=/home/srl/ros_ws/install_isolated -D 'depthai_DIR=$/home/srl/depthai-core/build/install/lib/cmake/depthai/' -G 'Unix Makefiles'

Command failed, exiting.

`
@saching13

saching13, 2021-04-02:
The command executed is still wrong. 

> srl@srl-HP-ProBook-440-G5:~/ros_ws$ catkin_make_isolated --cmake-args -D depthai_DIR=$/home/srl/depthai-core/build/install/lib/cmake/depthai/"

There should be no $ in the command.

ans-qureshi, 2021-04-02:
@saching13 oh crap, yuppp works! 💯 
Thanks for bearing me :)

ans-qureshi, 2021-04-15:
Hi! I'm getting following error after running command # 16 on ROS Melodic:

==> Processing catkin package: 'depthai_examples'
==> Building with env: '/home/srl/ros_ws/devel_isolated/depthai_bridge/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/srl/ros_ws/build_isolated/depthai_examples'
==> make -j8 -l8 in '/home/srl/ros_ws/build_isolated/depthai_examples'
[  0%] Built target vision_msgs_generate_messages_cpp
[  0%] Built target vision_msgs_generate_messages_eus
[  0%] Built target vision_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target vision_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target depthai_ros_msgs_generate_messages_eus
[  0%] Built target vision_msgs_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target depthai_ros_msgs_generate_messages_nodejs
[  0%] Built target depthai_ros_msgs_generate_messages_cpp
[  0%] Built target depthai_ros_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target depthai_ros_msgs_generate_messages_lisp
[ 30%] Built target depthai_examples
[ 40%] Built target rgb_stereo_node
[ 45%] Linking CXX executable /home/srl/ros_ws/devel_isolated/depthai_examples/lib/depthai_examples/rgb_subscriber_node
[ 55%] Built target rgb_node
[ 65%] Built target mobilenet_node
[ 75%] Built target stereo_node
[ 95%] Built target stereo_mimic_node
[ 95%] Built target nodelet_stereo
CMakeFiles/rgb_subscriber_node.dir/src/rgb_video_subscriber.cpp.o: In function `rgbCallback(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > > const&)':
rgb_video_subscriber.cpp:(.text+0x88): undefined reference to `cv::imshow(cv::String const&, cv::_InputArray const&)'
rgb_video_subscriber.cpp:(.text+0xb0): undefined reference to `cv::waitKey(int)'
collect2: error: ld returned 1 exit status
CMakeFiles/rgb_subscriber_node.dir/build.make:150: recipe for target '/home/srl/ros_ws/devel_isolated/depthai_examples/lib/depthai_examples/rgb_subscriber_node' failed
make[2]: *** [/home/srl/ros_ws/devel_isolated/depthai_examples/lib/depthai_examples/rgb_subscriber_node] Error 1
CMakeFiles/Makefile2:1173: recipe for target 'CMakeFiles/rgb_subscriber_node.dir/all' failed
make[1]: *** [CMakeFiles/rgb_subscriber_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'depthai_examples': 
  Command '['/home/srl/ros_ws/devel_isolated/depthai_bridge/env.sh', 'make', '-j8', '-l8']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/srl/ros_ws/build_isolated/depthai_examples && /home/srl/ros_ws/devel_isolated/depthai_bridge/env.sh make -j8 -l8

Command failed, exiting.

ans-qureshi, 2021-04-15:
Worked after adding following to CMakeList of depthai_examples:

```
include_directories(
#include
  ${catkin_INCLUDE_DIRS}
  ${OpenCV_INCLUDE_DIRS}
)
```

closed 2021-04-15 by ans-qureshi

Luxonis-Brandon, 2021-04-15:
Thanks for circling back!

saching13, 2021-04-15:
Thanks. 
Commenting include shouldn't fix the issue actually. I am working on opencv linking errors in `noetic-opencv-fix` will add the fix real soon..

saching13, 2021-04-16:
@ans-qureshi I used base ubuntu dockers and fixed this issue on both Ubuntu 18 and 20. you shouldn't experience this error anymore. Feel free to try it out and let me know if you have any feedback.