luxonis/depthai-ros #3: RLException

jonititan, 2021-04-07:
I followed the instructions on a blank 20.04 Ubuntu laptop which I had just installed Noetic onto in accordance with the [instructions.](http://wiki.ros.org/noetic/Installation/Ubuntu)
I then followed the instructions here but I get an exception when I attempt to launch the example.
if I run rospack list I can see the packages are not there.
running 
`source ~/ros_ws/devel_isolated/depthai_bridge/setup.bash`
results in the following depthai packages being found by rospack but launch still fails even from the launch directory.
depthai_bridge
depthai_ros_msgs

The exception I get is as follows
'RLException: [stereo_node.launch] is neither a launch file in package [depthai_examples] nor is [depthai_examples] a launch file name
The traceback for the exception was written to the log file'

jonititan, 2021-04-07:
Sorted.

I reran catkin_make_isolated and then after that a new setup.bash appeared in devel_isolated.
The command I used was 'catkin_make_isolated --cmake-args -D depthai_DIR=~/depthai_core/build/install/lib/cmake/depthai'
I ran
'source devel_isolated/setup.bash'  from ~/ros_ws
packages still didn't show in rospack but roslaunch now works.
It fails to find device but that is a different problem.

closed 2021-04-07 by jonititan

jonititan, 2021-04-07:
FYI the device connection was just needing to ensure the depthai python api is setup correctly to gen2 spec beforehand.

Luxonis-Brandon, 2021-04-07:
Thanks for making the issue and circling back with information on how to solve it.  This is super helpful for others who may run into it - so much appreciated.  :-)

saching13, 2021-04-07:
> FYI the device connection was just needing to ensure the depthai python api is setup correctly to gen2 spec beforehand.

Thanks for the comments @jonititan . I updated the path to a static path. 
can you elaborate on the quoted reply so that I can add steps to cover that. 
I think this is about the adding USB rules (Will add the link to that script here) is that the same?

jonititan, 2021-04-07:
What it turned out was I needed the depthai-python installed and configured.  Essentially my trouble shooting method was ensuring the gen2 examples worked correctly so that I could rule out faults in the camera or usb.

Once I had followed the instructions and verified that worked I retried and roslaunch completed correctly.

I didn't install opencv separately.

Does that make sense?

If needed I can compile a command history

saching13, 2021-04-07:
I think one is opencv which will be installed during installation of depthai-python. 
Second would be usb rules. 
Will add these two here. 

> If needed I can compile a command history

Thanks alot. if it's not too much work this would be helpful in checking if I am missing something.

jonititan, 2021-04-07:
Here is the slightly pruned command history from ~/.bash_history
Anything preceded by a # is a comment from me.

# first I installed ROS Noetic  http://wiki.ros.org/noetic/Installation/Ubuntu

sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
sudo apt update
sudo apt install ros-noetic-desktop-full
source /opt/ros/noetic/setup.bash
sudo apt install python3-rosdep python3-rosinstall python3-rosinstall-generator python3-wstool build-essential
sudo apt install python3-rosdep
sudo rosdep init
rosdep update
# then i followed the instructions to install depthai ros found here https://github.com/luxonis/depthai-ros  they have subsequently been revised

git clone --recursive https://github.com/luxonis/depthai-core.git --branch develop
cd depthai-core
mkdir build
cd build
cmake .. -D BUILD_SHARED_LIBS=ON
cmake --build . --parallel --config Release --target install
cd ~
mkdir -p ros_ws/src
cd ros_ws/src
git clone https://github.com/luxonis/depthai-ros.git --branch noetic-devel
git clone https://github.com/luxonis/depthai-ros-examples.git --branch noetic-devel
git clone https://github.com/ros-perception/vision_msgs.git --branch noetic-devel
cd ~/ros_ws
catkin_make_isolated --cmake-args -D depthai_DIR=<depthai-core insall directory>/lib/cmake/depthai
catkin_make_isolated --cmake-args -D depthai_DIR=depthai-core/lib/cmake/depthai
cd /home/jonititan/ros_ws/build_isolated/depthai_bridge && /home/jonititan/ros_ws/devel_isolated/depthai_ros_msgs/env.sh cmake /home/jonititan/ros_ws/src/depthai-ros/depthai_bridge -DCATKIN_DEVEL_PREFIX=/home/jonititan/ros_ws/devel_isolated/depthai_bridge -DCMAKE_INSTALL_PREFIX=/home/jonititan/ros_ws/install_isolated -D depthai_DIR=depthai-core/lib/cmake/depthai -G 'Unix Makefiles'
catkin_make_isolated --cmake-args -D depthai_DIR=depthai-core/lib/cmake/depthai
cd ~
cd ros_ws
source /opt/ros/noetic/setup.bash
catkin_make_isolated --cmake-args -D depthai_DIR=depthai-core/lib/cmake/depthai

% this command gave the error message that the path was wrong and led to much screwing around trying to work it out.  I've omitted that.  

catkin_make_isolated --cmake-args -D depthai_DIR=~/depthai-core/build/install/lib/cmake/depthai

roslaunch depthai_examples stereo_node.launch && roslaunch
# then roslaunch wouldn't launch...
# cue more screwing around until i worked out like with much of ROS i needed to source a setup.bash file.  Sadly this wasn't listed and I had to find the right one.

source ~/ros_ws/devel_isolated/setup.bash

roslaunch depthai_examples stereo_node.launch 

# then it launched but there was a connection error to the camera.  Cue more screwing around.  I decided to try the gen2 examples to test the connection. This involved installing the python api and wondering if that was a required dependency for the ros package.... ?

cd ../ # getting back to home directory  I then tried to install via pip as instructed in https://docs.luxonis.com/projects/api/en/latest/ but this was premature...
python3 -m pip install --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/ depthai==2.1.0.0.dev+e5a09bf1a442d4be0cc2c8f4bb06b47e1c661ead #discovered pip wasn't installed...  blank ubuntu install...
sudo apt install python3-pip
python3 -m pip install --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/ depthai==2.1.0.0.dev+e5a09bf1a442d4be0cc2c8f4bb06b47e1c661ead
roslaunch depthai_examples stereo_node.launch && roslaunch depthai_examples stereo_nodelet.launch && roslaunch depthai_examples rgb_publisher.launch  # still didn't work
# I followed the instructions for the gen2 examples https://docs.luxonis.com/projects/api/en/latest/install/

python3 -m pip install -U pip
python3 -m pip install opencv-python
python3 -m pip install -U --force-reinstall depthai

# i forgot about the dependencies so i did that bit...
sudo wget -qO- http://docs.luxonis.com/_static/install_dependencies.sh | bash
#I downloaded and unzipped the python api manually and ran this command from the folder the examples were within. https://github.com/luxonis/depthai-python/tree/main/examples
python3 install_requirements.py
python3 08_rgb_mobilenet.py # this worked
cd ../ #getting back to home directory
cd ros_ws
source ~/ros_ws/devel_isolated/setup.bash
roslaunch depthai_examples stereo_nodelet.launch #it worked this time

saching13, 2021-04-07:
Thanks a lot for the detailed logs and feedback.

Looks like `python3 -m pip install opencv-python`
and `sudo wget -qO- http://docs.luxonis.com/_static/install_dependencies.sh | bash`
fixed the issue for you. 
will add a reference to those for first-time users.

jonititan, 2021-04-07:
yeah I think so.  I get there in the end but as you can probably see I have to do a lot of wandering to work it out.  Still learning.
I think it's because there are assumed dependencies prior to installing depthai-ros.  I naively didn't anticipate them.  Sorry about the formatting.

saching13, 2021-04-07:
Yes, It was my bad that I made that assumption. Will fix it.

Luxonis-Brandon, 2021-04-07:
Sweet, thanks both!

jonititan, 2021-04-07:
No worries.  This is cool stuff but it's all advancing so its no wonder the documentation lags.  What there was already still helped me to work out what was going on.  Thanks for your hard work on this.