luxonis/depthai-ros #4: Some issues encountered with the stereo_nodelet example

jonititan, 2021-04-09:
ROS report from using stereo_nodelet.launch

Warning message during launch.  Implies config error?

[ WARN] [1617956295.079873425]: [stereo] does not match name right in file /home/jonititan/ros_ws/src/depthai-ros-examples/depthai_examples/params/camera/right.yaml

Strange Error.  When I record everything a rosbag I get image format errors continuously and an occasional warning about sync.

[ERROR] [1617954294.986707649]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)
[ERROR] [1617954294.987537467]: cv_bridge exception: '[32FC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1617954295.478277395]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)
[ERROR] [1617954295.479139603]: cv_bridge exception: '[32FC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1617954295.967646872]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)
[ERROR] [1617954295.968534650]: cv_bridge exception: '[32FC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1617954296.450010437]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)
[ERROR] [1617954296.450878760]: cv_bridge exception: '[32FC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1617954296.951450260]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)
[ERROR] [1617954296.952243234]: cv_bridge exception: '[32FC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ WARN] [1617954297.221679545]: [image_transport] Topics '/stereo_publisher/stereo/image' and '/stereo_publisher/stereo/camera_info' do not appear to be synchronized. In the last 10s:
	Image messages received:      20
	CameraInfo messages received: 73
	Synchronized pairs:           20

I haven't yet had time to go through topics one by one to work out which cause this issue.  Hopefully I will soon.

jonititan, 2021-04-09:
[image]

jonititan, 2021-04-09:
Obviously it was theora and the compressed images causing the Errors.
When I excldue this streams from the recording using
rosbag record -a -x "/stereo_publisher/stereo/image/theora|/stereo_publisher/stereo/image/theora/(.*)|/stereo_publisher/stereo/image/compressed|/stereo_publisher/stereo/image/compressed/(.*)" 

I no longer get the error.  However I do still get the warning about 
Topics '/stereo_publisher/stereo/image' and '/stereo_publisher/stereo/camera_info' do not appear to be synchronized.

Luxonis-Brandon, 2021-04-09:
CC: @saching13

saching13, 2021-04-09:
hi @jonititan 

`roslaunch depthai_examples stereo_nodelet.launch`

I ran the above command and I didn't get any synchronization issues. 
And as the matter of fact. It is highly unlikely that it happens since I am making sure `camera_info` and its respective stream from the device is using the same header.
And as of the warning printed you can ignore that. it is because the topic ns is different than that of the file name.

[image]

and as of this error

> [ERROR] [1617954295.479139603]: cv_bridge exception: '[32FC1] is not a color format. but [bgr8] is. The conversion does not make sense'

As of this example where is this commming from and what is theora. I have never seen that name in our examples.

saching13, 2021-04-09:
I think I know what is happening here. 

Our device is publishing depth_map at `/stereo_publisher/stereo/depth` and depth is in millimeters. And as of `deptha_image_top_pointcloud_nodelet` of `depth_image_proc` needs it to be in meters. 

So they provided a `depth_image_convertion_nodelet` which subscribes to `/stereo_publisher/stereo/depth`, converts it into meters and republished at `/stereo_publisher/stereo/depth`. And when this happens it looks like they are adding a new timestamp to that. 
and if the this conversion process is show due to processing power limitation on the host it will lead to synchronization issue with the `camera_info` which is published along with the image from the device.

jonititan, 2021-04-09:
That makes sense.  I was running it on a laptop.  
I did a trial run today so I can gather enough data to really go over different algorithms to use.  I'll go through and try to identify where those other bits are coming from.

I haven't modified it from your example however and I had nothing else running in ROS.

saching13, 2021-04-09:
I checked `depth_image_proc` and it's source code and looks like that's the culprit here.
The usual norm is that when you modify an image and re-publishing that image then you also should handle publishing its `camera_info` also which with appropriate timestep which is not done by their library hence synchronization issue when there is a delay in processing the incoming messages.

jonititan, 2021-04-09:
Here's the output from rostopic list -v

Published topics:
 * /rosout_agg [rosgraph_msgs/Log] 1 publisher
 * /rosout [rosgraph_msgs/Log] 5 publishers
 * /tf [tf2_msgs/TFMessage] 1 publisher
 * /tf_static [tf2_msgs/TFMessage] 1 publisher
 * /stereo_publisher/stereo/image/compressedDepth [sensor_msgs/CompressedImage] 1 publisher
 * /stereo_publisher/stereo/image/compressedDepth/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /stereo_publisher/stereo/image/compressedDepth/parameter_updates [dynamic_reconfigure/Config] 1 publisher
 * /stereo_publisher/stereo/image/compressed [sensor_msgs/CompressedImage] 1 publisher
 * /stereo_publisher/stereo/image/compressed/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /stereo_publisher/stereo/image/compressed/parameter_updates [dynamic_reconfigure/Config] 1 publisher
 * /stereo_publisher/stereo/image [sensor_msgs/Image] 1 publisher
 * /stereo_publisher/stereo/image/theora [theora_image_transport/Packet] 1 publisher
 * /stereo_publisher/stereo/image/theora/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /stereo_publisher/stereo/image/theora/parameter_updates [dynamic_reconfigure/Config] 1 publisher
 * /nodelet_manager/bond [bond/Status] 3 publishers
 * /stereo/points [sensor_msgs/PointCloud2] 1 publisher
 * /stereo_publisher/left/image [sensor_msgs/Image] 1 publisher
 * /stereo_publisher/left/camera_info [sensor_msgs/CameraInfo] 1 publisher
 * /stereo_publisher/right/image [sensor_msgs/Image] 1 publisher
 * /stereo_publisher/right/camera_info [sensor_msgs/CameraInfo] 1 publisher
 * /stereo_publisher/stereo/depth [sensor_msgs/Image] 1 publisher
 * /stereo_publisher/stereo/camera_info [sensor_msgs/CameraInfo] 1 publisher

Subscribed topics:
 * /rosout [rosgraph_msgs/Log] 1 subscriber
 * /joint_states [sensor_msgs/JointState] 1 subscriber
 * /nodelet_manager/bond [bond/Status] 3 subscribers

Also here's the graph

[image]

The theora and compressed image related topics don't seem to appear in the graph so maybe they are inactive?  If so I wonder why they are causing problems for Rosbag record or where they come from in the first place.

saching13, 2021-04-09:
they seem to be inactive. they are created by `imageTransport::Publisher` in `depth_image_proc`.

[depth_image_proc](http://wiki.ros.org/depth_image_proc) has
`depth_image_proc/convert_metric` and `depth_image_proc/point_cloud_xyz` nodelets which creates them using ImageTransport. 

You can take a look at that package to see in more detail about what is inactive.

jonititan, 2021-04-09:
Will do.  It didn't end up being a problem in the end as I made my own modded version which concentrated on getting the rgb stream at full resolution.
I'll dig further into this on my own time but since it seems most of the issue was hardware limitations my end perhaps we should just close this?

saching13, 2021-04-09:
Sure. I will close this. 
And this is most of the `depth_image_proc` package limitation than your hardware limitation. 

Thanks a lot for the feedbacks. Feel free to add any issues you face.

closed 2021-04-09 by saching13