luxonis/depthai-ros #5: YAML-cpp linker issue while installing using catkin_make

botmayank, 2021-04-13:
Hey, I'm getting a linker error in the final step while trying to build the depthai_examples.

```
catkin_make_isolated --cmake-args -D depthai_DIR=/media/mayank/Data/999_Misc/6_OAKD/depthai-core/build/install/lib/cmake/depthai -DPYTHON_EXECUTABLE=/usr/bin/python3
Base path: /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws
Source space: /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/src
Build space: /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/build_isolated
Devel space: /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel_isolated
Install space: /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/install_isolated
Additional CMake Arguments: -D depthai_DIR=/media/mayank/Data/999_Misc/6_OAKD/depthai-core/build/install/lib/cmake/depthai -DPYTHON_EXECUTABLE=/usr/bin/python3
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 4 packages in topological order:
~~  - vision_msgs
~~  - depthai_ros_msgs
~~  - depthai_bridge
~~  - depthai_examples
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing catkin package: 'vision_msgs'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/build_isolated/vision_msgs'
==> make -j12 -l12 in '/media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/build_isolated/vision_msgs'
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target _vision_msgs_generate_messages_check_deps_VisionInfo
[  0%] Built target _vision_msgs_generate_messages_check_deps_Detection2DArray
[  0%] Built target _vision_msgs_generate_messages_check_deps_Detection3D
[  0%] Built target _vision_msgs_generate_messages_check_deps_Detection2D
[  0%] Built target _vision_msgs_generate_messages_check_deps_Detection3DArray
[  0%] Built target _vision_msgs_generate_messages_check_deps_BoundingBox3D
[  0%] Built target _vision_msgs_generate_messages_check_deps_ObjectHypothesisWithPose
[  0%] Built target _vision_msgs_generate_messages_check_deps_BoundingBox2D
[  0%] Built target _vision_msgs_generate_messages_check_deps_Classification3D
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target _vision_msgs_generate_messages_check_deps_ObjectHypothesis
[  0%] Built target _vision_msgs_generate_messages_check_deps_BoundingBox3DArray
[  0%] Built target _vision_msgs_generate_messages_check_deps_Classification2D
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_lisp
[ 30%] Built target vision_msgs_generate_messages_eus
[ 40%] Built target vision_msgs_generate_messages_nodejs
[ 59%] Built target vision_msgs_generate_messages_cpp
[ 79%] Built target vision_msgs_generate_messages_lisp
[100%] Built target vision_msgs_generate_messages_py
[100%] Built target vision_msgs_generate_messages
<== Finished processing package [1 of 4]: 'vision_msgs'

==> Processing catkin package: 'depthai_ros_msgs'
==> Building with env: '/media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel_isolated/vision_msgs/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/build_isolated/depthai_ros_msgs'
==> make -j12 -l12 in '/media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/build_isolated/depthai_ros_msgs'
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target vision_msgs_generate_messages_cpp
[  0%] Built target vision_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_py
[... 192 lines cut ...]
          /opt/yaml-cpp/
          ${YAMLCPP_DIR}/include/)

# find the yaml-cpp library
find_library(YAMLCPP_LIBRARY
             NAMES ${YAMLCPP_STATIC} yaml-cpp
             PATH_SUFFIXES lib64 lib
             PATHS ~/Library/Frameworks
                    /Library/Frameworks
                    /usr/local
                    /usr
                    /sw
                    /opt/local
                    /opt/csw
                    /opt
                    ${YAMLCPP_DIR}/lib)

# handle the QUIETLY and REQUIRED arguments and set YAMLCPP_FOUND to TRUE if all listed variables are TRUE
include(FindPackageHandleStandardArgs)
FIND_PACKAGE_HANDLE_STANDARD_ARGS(YAMLCPP DEFAULT_MSG YAMLCPP_INCLUDE_DIR YAMLCPP_LIBRARY)
mark_as_advanced(YAMLCPP_INCLUDE_DIR YAMLCPP_LIBRARY)
```
It even seems like CMake initially finds YAML, but the linker still complains. 
`-- Found YAMLCPP: /usr/local/include `
I even tried cleaning up the build and devel directories but still the same issue persists.

saching13, 2021-04-13:
> `/usr/bin/ld: warning: libopencv_imgcodecs.so.4.2, needed by /opt/ros/noetic/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.4.5`

You seem to be having two versions of OpenCV which is conflicting while linking.

botmayank, 2021-04-13:
That's a warning which I can fix by giving an explicit path to my opencv install. The main ld error IMO is coming from the undefined symbols for YAML methods.

saching13, 2021-04-13:
> `/usr/bin/ld: /opt/ros/noetic/lib/libcamera_calibration_parsers.so: undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder>)'`

Our packages don't use yaml anywhere. the error is coming from the openCV. and from your setup you are getting only linking error it means that you are able to find the include files but fails to find the build lib file of YAMPCPP. 

Can you confirm if using openCV 4 in other packages also creates this issue ? or is it specific to depthai packages ? 
And can you share your setup info too so that I can try to recreate it.

saching13, 2021-04-13:
I made opencv to import only specific componenets which are needed and might avoid this issue. 
can you checkout `noetic-opencv-fix` from both `depthai-ros` and  `depthai-ros-examples` and see if it works.
I will try to test it in my other systems in the maintime.

saching13, 2021-04-16:
@botmayank I updated the OpenCV dependency. Can you try it and let me know if the new update fixes the issue.
Thanks.

botmayank, 2021-04-24:
Hi @saching13 I'm able to successfully build depthai_bridge and depthai_ros_msgs, vision_msgs. Depthai_examples still fails (even noetic devel with the merged noetic-opencv-fix)

I had Opencv 4.2 and 4.5 both installed. I've removed 4.5 since ROS seemed to be dependent on 4.2. I'm still getting the linker errors for linking:

```
[ 87%] Built target nodelet_stereo
[ 89%] Linking CXX executable /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel/lib/depthai_examples/stereo_mimic_node
[ 91%] Linking CXX executable /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel/lib/depthai_examples/mobilenet_node
[ 91%] Linking CXX executable /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel/lib/depthai_examples/rgb_node
[ 92%] Built target rgb_subscriber_node
[ 93%] Linking CXX executable /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel/lib/depthai_examples/rgb_stereo_node
[ 94%] Linking CXX executable /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel/lib/depthai_examples/stereo_node
/usr/bin/ld: /opt/ros/noetic/lib/libcamera_calibration_parsers.so: undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder>)'
collect2: error: ld returned 1 exit status
make[2]: *** [depthai-ros-examples/depthai_examples/CMakeFiles/stereo_node.dir/build.make:241: /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel/lib/depthai_examples/stereo_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:2850: depthai-ros-examples/depthai_examples/CMakeFiles/stereo_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: /opt/ros/noetic/lib/libcamera_calibration_parsers.so: undefined reference to `YAML::detail::/node_data:usr:convert_to_map/(binstd/:ld::shared_ptr </YAMLopt:/:rosdetail/:noetic:/memory_holderlib>/)libcamera_calibration_parsers.so':
 undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder>)'collect2: error: ld returned 1 exit status

collect2: error: ld returned 1 exit status
/usr/bin/ld: /opt/ros/noetic/lib/libcamera_calibration_parsers.so: undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder>)'
collect2: error: ld returned 1 exit status
make[2]: *** [depthai-ros-examples/depthai_examples/CMakeFiles/rgb_node.dir/build.make:241: /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel/lib/depthai_examples/rgb_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:3102: depthai-ros-examples/depthai_examples/CMakeFiles/rgb_node.dir/all] Error 2
make[2]: *** [depthai-ros-examples/depthai_examples/CMakeFiles/mobilenet_node.dir/build.make:241: /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel/lib/depthai_examples/mobilenet_node] Error 1
make[2]: *** [depthai-ros-examples/depthai_examples/CMakeFiles/rgb_stereo_node.dir/build.make:241: /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel/lib/depthai_examples/rgb_stereo_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:2976: depthai-ros-examples/depthai_examples/CMakeFiles/rgb_stereo_node.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2913: depthai-ros-examples/depthai_examples/CMakeFiles/mobilenet_node.dir/all] Error 2
/usr/bin/ld: /opt/ros/noetic/lib/libcamera_calibration_parsers.so: undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder>)'
collect2: error: ld returned 1 exit status
make[2]: *** [depthai-ros-examples/depthai_examples/CMakeFiles/stereo_mimic_node.dir/build.make:241: /media/mayank/Data/999_Misc/6_OAKD/depthai_ros_ws/devel/lib/depthai_examples/stereo_mimic_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:3039: depthai-ros-examples/depthai_examples/CMakeFiles/stereo_mimic_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j12 -l12" failed
```

saching13, 2021-04-24:
Thanks for checking. 
Can you share your system information and your steps so that I can try to recreate this ?

botmayank, 2021-04-25:
Yes I deleted build and devel to retry.

The command I'm using to build (since other packages built successfully...):
`catkin_make --only-pkg-with-deps depthai_examples --cmake-args -D depthai_DIR=<depthai_path>/depthai-core/build/install/lib/cmake/depthai -DPYTHON_EXECUTABLE=/usr/bin/python3`

Things I've tried:
1. Adding the snippet in the prev comment for CMake to find libyaml
2. Ensuring LD_LIBRARY_PATH env var has /usr/local/lib where libyaml-cpp.so is present
3. Manually adding cpp compiler flags with paths in CMakeLists.txt:
`add_compile_options(-std=c++14 -lyaml-cpp -L/usr/local/lib -I/usr/local/include)`
4. Rebuilding libyaml from source from source and installing both .so and .a versions from:
https://github.com/jbeder/yaml-cpp

What all system info do you want? 

1. ROS version: Noetic
2. OS Version: Ubuntu 20.04.2 LTS
3. Arch: x86_64 (output of uname -a: 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux)

Additional sanity checks:

So the complaining library is libcamera_calibration_parsers.so and the offending symbol is YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder>)

When I check:
`readelf --dynamic /opt/ros/noetic/lib/libcamera_calibration_parsers.so | grep yaml` it correctly shows:
` 0x0000000000000001 (NEEDED)             Shared library: [libyaml-cpp.so.0.6]`

`nm --print-file-name --dynamic /usr/local/lib/libyaml-cpp.so.0.6 | grep convert_to_map` also shows:
`/usr/local/lib/libyaml-cpp.so.0.6:00000000000bcd3a T _ZN4YAML6detail9node_data14convert_to_mapERKSt10shared_ptrINS0_13memory_holderEE
` 
If I check ldconfig:

`sudo ldconfig -p | grep yaml`
Output:
`sudo ldconfig -p | grep yaml
	libyaml-0.so.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libyaml-0.so.2
	libyaml-cpp.so.0.6 (libc6,x86-64) => /usr/local/lib/libyaml-cpp.so.0.6
	libyaml-cpp.so.0.6 (libc6,x86-64) => /lib/x86_64-linux-gnu/libyaml-cpp.so.0.6
	libyaml-cpp.so (libc6,x86-64) => /usr/local/lib/libyaml-cpp.so
	libyaml-cpp.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libyaml-cpp.so
`
As far as I can tell, I seem to have the library on my system which even has this symbol (unless there's a minor difference in definition I'm missing). The issue seems to be that `ld` still doesn't see it although I've tried every which way to tell it where the library is.

botmayank, 2021-04-25:
Additionally, 

```
sudo ldconfig -p | grep opencv
[sudo] password for mayank: 
	libopencv_xphoto.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_xphoto.so.4.2
	libopencv_xphoto.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_xphoto.so
	libopencv_xobjdetect.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_xobjdetect.so.4.2
	libopencv_xobjdetect.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_xobjdetect.so
	libopencv_ximgproc.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_ximgproc.so.4.2
	libopencv_ximgproc.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_ximgproc.so
	libopencv_viz.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_viz.so.4.2
	libopencv_viz.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_viz.so
	libopencv_videostab.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_videostab.so.4.2
	libopencv_videostab.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_videostab.so
	libopencv_videoio.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_videoio.so.4.2
	libopencv_videoio.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_videoio.so
	libopencv_video.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_video.so.4.2
	libopencv_video.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_video.so
	libopencv_tracking.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_tracking.so.4.2
	libopencv_tracking.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_tracking.so
	libopencv_text.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_text.so.4.2
	libopencv_text.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_text.so
	libopencv_surface_matching.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_surface_matching.so.4.2
	libopencv_surface_matching.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_surface_matching.so
	libopencv_superres.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_superres.so.4.2
	libopencv_superres.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_superres.so
	libopencv_structured_light.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_structured_light.so.4.2
	libopencv_structured_light.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_structured_light.so
	libopencv_stitching.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_stitching.so.4.2
	libopencv_stitching.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_stitching.so
	libopencv_stereo.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_stereo.so.4.2
	libopencv_stereo.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_stereo.so
	libopencv_shape.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_shape.so.4.2
	libopencv_shape.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_shape.so
	libopencv_saliency.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_saliency.so.4.2
	libopencv_saliency.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_saliency.so
	libopencv_rgbd.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_rgbd.so.4.2
	libopencv_rgbd.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_rgbd.so
	libopencv_reg.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_reg.so.4.2
	libopencv_reg.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_reg.so
	libopencv_quality.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_quality.so.4.2
	libopencv_quality.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_quality.so
	libopencv_plot.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_plot.so.4.2
	libopencv_plot.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_plot.so
	libopencv_photo.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_photo.so.4.2
	libopencv_photo.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_photo.so
	libopencv_phase_unwrapping.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_phase_unwrapping.so.4.2
	libopencv_phase_unwrapping.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_phase_unwrapping.so
	libopencv_optflow.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_optflow.so.4.2
	libopencv_optflow.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_optflow.so
	libopencv_objdetect.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_objdetect.so.4.2
	libopencv_objdetect.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_objdetect.so
	libopencv_ml.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_ml.so.4.2
	libopencv_ml.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_ml.so
	libopencv_line_descriptor.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_line_descriptor.so.4.2
	libopencv_line_descriptor.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_line_descriptor.so
	libopencv_imgproc.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_imgproc.so.4.2
	libopencv_imgproc.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_imgproc.so
	libopencv_imgcodecs.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_imgcodecs.so.4.2
	libopencv_imgcodecs.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_imgcodecs.so
	libopencv_img_hash.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_img_hash.so.4.2
	libopencv_img_hash.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_img_hash.so
	libopencv_highgui.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_highgui.so.4.2
	libopencv_highgui.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_highgui.so
	libopencv_hfs.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_hfs.so.4.2
[... cut ...]
	libopencv_bioinspired.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_bioinspired.so.4.2
	libopencv_bioinspired.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_bioinspired.so
	libopencv_bgsegm.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_bgsegm.so.4.2
	libopencv_bgsegm.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_bgsegm.so
	libopencv_aruco.so.4.2 (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_aruco.so.4.2
	libopencv_aruco.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libopencv_aruco.so

```

saching13, 2021-04-25:
Thanks. I will try to recreate it and see if I can fix on our side. 
And cc: @themarpe in case if you have any suggestions on this.

themarpe, 2021-04-26:
@botmayank @saching13 
Seems like the signature of that function changed.

Required:
```
YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder>)
```

Installed:
```
YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder> const&) 
```

This is the commit that changed the signature: https://github.com/jbeder/yaml-cpp/commit/0b0bf35c48806487958cd375782767cad027b674#diff-2f31b22281fe05e5b3bc2d4106bac98b28cf93d96c10695b857824672f67f23f  [Jun 16, 2020]

I think that the latest release (not latest `master` branch) https://github.com/jbeder/yaml-cpp/releases/tag/yaml-cpp-0.6.3
should not have this yet applied. [Sep 25, 2019]

TLDR: Remove all yaml-cpp related files in `/usr/local` (includes, libraries, ...) and rebuild and install the above release (`yaml-cpp-0.6.3`).

saching13, 2021-04-26:
@botmayank  Looks like your issue is coming from [camera_info_manager](http://wiki.ros.org/camera_info_manager) where it is trying to use `camera_calibration_parser` link [here](https://github.com/ros-perception/image_common/blob/3eed8b5cf542d356051094535916ca180b2d3e6c/camera_info_manager/CMakeLists.txt#L9)

Looks like your `ros-perception` packages are somehow messed up internal linking. Can you try reinstalling those packages properly. That should fix your issue without any extra work.

botmayank, 2021-04-29:
@themarpe nice catch! Thanks a ton that was it :partying_face: 
@saching13 Since I installed ROS using `sudo apt install ros-noetic-desktop-full`, if I try to remove `ros-noetic-perception` using apt, it'll uninstall ROS desktop too which I want to avoid. I also think that won't fix it. 

For me, the issue is resolved by building and installing yaml-cpp-0.6.3. 

TL;DR fix:
1. Remove libyaml-cpp from your system. For me that meant:
```
sudo rm /usr/local/lib/libyaml-cpp*
sudo rm /lib/x86_64-linux-gnu/libyaml-cpp*
```
2. Clone release 0.6.3 for libyaml-cpp:
`git clone https://github.com/jbeder/yaml-cpp.git --branch yaml-cpp-0.6.3`

3. Build and install libyaml-cpp as per https://github.com/jbeder/yaml-cpp/blob/master/install.txt
```
mkdir build && cd build
cmake ..
make -j12
sudo make install
```
4. Build depthai_ros_examples:
`catkin_make --only-pkg-with-deps depthai_examples --cmake-args -D depthai_DIR=<depthai_path>/depthai-core/build/install/lib/cmake/depthai -DPYTHON_EXECUTABLE=/usr/bin/python3`

closed 2021-04-29 by botmayank