luxonis/depthai-ros #7: Spatial Detection Example Problem

luiscastro1995, 2021-04-20:
I am trying to adapt the **mobile_publisher** example from the **depthai-ros-examples** to work with the **yolov4** network publishing the spatial coordinates (X, Y, Z) of each detection. I have come up with the following codes: https://drive.google.com/drive/folders/1DqpVrxvffyWEZXiJ9GvtiCxAthoPASu6?usp=sharing
If you want to replicate my results please place the following files in the respective folders:
- yolov4_spatial_publisher.cpp -> ~/ros_ws/ros_depthai_ws/src/depthai-ros-examples/depthai_examples/src
- yolov4_spatial_pipeline.cpp -> ~/ros_ws/ros_depthai_ws/src/depthai-ros-examples/depthai_examples/lib
- yolov4_spatial_publisher.launch -> ~/ros_ws/ros_depthai_ws/src/depthai-ros-examples/depthai_examples/launch
- yolov4_spatial_pipeline.hpp -> ~/ros_ws/ros_depthai_ws/src/depthai-ros-examples/depthai_examples/include/depthai_examples
- Copy "tiny-yolo-v4_openvino_2021.2_6shave.blob" to somewhere, Update the path to this file in line 29 of code yolov4_spatial_publisher.cpp

To compile I did **catkin_make_isolated --cmake-args -Ddepthai_DIR=~/depthai-core/build/install/lib/cmake/depthai** since I am using ROS Melodic

I am able to compile and run the code. But when I inspect the rbg images from the camera, using rqt_img_viewer plugin, it outputs 1 or 2 frames then stops completely. And there is nothing being published on the topic of the NN detection.
I have tried the same model without using the spatial info and it worked fine. So, I guess the main problem should be on **yolov4_spatial_pipeline.cpp** where I define the pipeline or in **yolov4_spatial_publisher.cpp** where I create the topics and publishers.

Any suggestions are more than welcome!

saching13, 2021-04-20:
Thanks for sharing the code. I will try the code at the earliest and check whats the issue and get back to you.

luiscastro1995, 2021-04-21:
Thank you for you time!
Forgot to tell that you need to add this line to the **CMakeLists.txt**:
- dai_add_node(yolov4_spatial_node src/yolov4_spatial_publisher.cpp)

saching13, 2021-05-11:
Hi,
Sorry for the long wait. Have been busy with lot of things.

Here is the sample result I got with some modifications to your code. 

```
header: 
  seq: 346
  stamp: 
    secs: 1620761857
    nsecs:  12942470
  frame_id: "oak_rgb_camera_optical_frame"
detections: 
  - 
    results: 
      - 
        id: 0
        score: 0.6420607566833496
    bbox: 
      center: 
        x: 174.5
        y: 265.5
        theta: 0.0
      size_x: 365.0
      size_y: 295.0
    position: 
      x: -0.0558454692363739
      y: -0.12286003679037094
      z: 1.2330000400543213
    is_tracking: False
    tracking_id: ''
---
header: 
  seq: 347
  stamp: 
    secs: 1620761857
    nsecs:  14133957
  frame_id: "oak_rgb_camera_optical_frame"
detections: 
  - 
    results: 
      - 
        id: 0
        score: 0.5141294002532959
    bbox: 
      center: 
        x: 174.0
        y: 265.5
        theta: 0.0
      size_x: 364.0
      size_y: 293.0
    position: 
      x: -0.05792438983917236
      y: -0.12136537581682205
      z: 1.218000054359436
    is_tracking: False
    tracking_id: ''
```

luiscastro1995, 2021-05-13:
Sorry for the late reply.
Thank you very much for your time and the great news!
If you could add that to the examples would be perfect, thanks!

closed 2021-05-17 by luiscastro1995