luxonis/depthai-ros #8: catkin compilation error in SpatialDetectionConverter.cpp:69:25

h3ct0r, 2021-05-03:
Hi,

I'm using Ubuntu 20.04, with ROS Noetic. All packages installed without problems as stated by the tutorial (including the required `depthai-core`), however, at the final step for compiling the `depthai_bridge` with `catkin_make`/`catkin build` gives an error that does not seems to be from library issues:

```
/home/h3ct0r/catkin_ws_depthai/src/luxonis/depthai-ros/depthai_bridge/src/SpatialDetectionConverter.cpp: In member function ‘void dai::rosBridge::SpatialDetectionConverter::toRosMsg(std::shared_ptr<dai::SpatialImgDetections>, depthai_ros_msgs::SpatialDetectionArray&)’:
/home/h3ct0r/catkin_ws_depthai/src/luxonis/depthai-ros/depthai_bridge/src/SpatialDetectionConverter.cpp:69:25: error: cannot convert ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’} to ‘vision_msgs::ObjectHypothesis_<std::allocator<void> >::_id_type’ {aka ‘long int’} in assignment
   69 |           std::to_string(inNetData->detections[i].label);
      |           ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                         |
      |                         std::string {aka std::__cxx11::basic_string<char>}
/home/h3ct0r/catkin_ws_depthai/src/luxonis/depthai-ros/depthai_bridge/src/ImgDetectionConverter.cpp: In member function ‘void dai::rosBridge::ImgDetectionConverter::toRosMsg(std::shared_ptr<dai::ImgDetections>, vision_msgs::Detection2DArray&)’:
/home/h3ct0r/catkin_ws_depthai/src/luxonis/depthai-ros/depthai_bridge/src/ImgDetectionConverter.cpp:69:25: error: cannot convert ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’} to ‘vision_msgs::ObjectHypothesisWithPose_<std::allocator<void> >::_id_type’ {aka ‘long int’} in assignment
   69 |           std::to_string(inNetData->detections[i].label);
      |           ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                         |
      |                         std::string {aka std::__cxx11::basic_string<char>}
make[2]: *** [luxonis/depthai-ros/depthai_bridge/CMakeFiles/depthai_bridge.dir/build.make:89: luxonis/depthai-ros/depthai_bridge/CMakeFiles/depthai_bridge.dir/src/SpatialDetectionConverter.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/h3ct0r/catkin_ws_depthai/src/luxonis/depthai-ros/depthai_bridge/src/ImageConverter.cpp: In member function ‘cv::Mat dai::rosBridge::ImageConverter::rosMsgtoCvMat(sensor_msgs::Image&)’:
/home/h3ct0r/catkin_ws_depthai/src/luxonis/depthai-ros/depthai_bridge/src/ImageConverter.cpp:247:1: warning: control reaches end of non-void function [-Wreturn-type]
  247 | }
      | ^
make[2]: *** [luxonis/depthai-ros/depthai_bridge/CMakeFiles/depthai_bridge.dir/build.make:76: luxonis/depthai-ros/depthai_bridge/CMakeFiles/depthai_bridge.dir/src/ImgDetectionConverter.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2554: luxonis/depthai-ros/depthai_bridge/CMakeFiles/depthai_bridge.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j16 -l16" failed
```

Any clues on how to solve this?

saching13, 2021-05-03:
`vision_msgs` package changed there data_structure which lead to this failue. 
it's fixed now. Make a pull once and build it again. 
Thanks

h3ct0r, 2021-05-03:
Thanks, its working now!

closed 2021-05-03 by h3ct0r