luxonis/depthai-ros #9: Support for Image Transport

berickson, 2021-05-04:
I started playing with the examples, and it looks like the ros endpoints for camera images only support the raw format.  I was previously publishing images with python and could see a huge performance improvement for remote subscribers using compressed images.

It looks like the recommended way to do this is to use image_transport (see http://wiki.ros.org/image_transport).  That adds support for multiple publish endpoints, compressed, Theora, etc.

Do you support / have any plans to support this?  It looks like it might require BridgePublisher specializations for sensor_msgs::Image.  I might try to code it up in a fork if you are interested.

saching13, 2021-05-05:
The reason why I didn't use it is because it needed specialization. 
And I left `NV12` to be published as it is to provide backward compatibility to depthai data structures. 
Wouldn't it be easier to use a nodelet to republish using `image_transport` ?

berickson, 2021-05-05:
A nodelet would probably be a great way to do it.