luxonis/depthai-ros #10: RGB Point Cloud

DaniGarciaLopez, 2021-05-07:
Hi!

I managed to launch the example 'roslaunch depthai_examples stereo_nodelet.launch' and I can visualize the point cloud in rviz.

How to visualize the point cloud with rgb information? I don't know if this feature is already done, I couldn't find the way to show it.

Thanks!

saching13, 2021-05-07:
> How to visualize the point cloud with rgb information? I don't know if this feature is already done, I couldn't find the way to show it.

Depth-RGB alignment was recently added in DepthAI (Only for the case when RGB is in the center of the device) for now. 
An example on this has not been created in ROS yet.

DaniGarciaLopez, 2021-05-07:
Thanks for the reply! Looking forward to have an example in ROS.

martinerk0, 2021-12-06:
Looking forward for this too. Also, why is the pointcloud so noisy?

Luxonis-Brandon, 2021-12-06:
Sorry about the trouble.  Likely we have incorrect defaults set.  Discussing internally now.

saching13, 2021-12-06:
@martinerk0  can you share a screenshot of it please ?
And I modified the confidence and LRThreshold too. And if the newly set parameters still feel like noisy for your usecase use the following parameters to tweak the confidence. 
` roslaunch depthai_examples stereo_nodelet.launch LRchecktresh:=5 confidence:=200` 

[image]

saching13, 2021-12-06:
[image]

Serafadam, 2022-11-07:
Closing due to inactivity, if you think this should remain open, please tag me in a comment here.

closed 2022-11-07 by Serafadam