luxonis/depthai-ros #11: ROS2 installation errors

luiscastro1995, 2021-05-17:
Hi @saching13!
After using the depthai-ros-examples in ROS Melodic, I would like to move on to ROS2. I was trying to install the depthai-ros packages in my ROS2 Eloquent workspace but it fails with errors like the kind:

Command run: colcon build --packages-select depthai_bridge depthai_ros_msgs
Log:
```bash
Starting >>> depthai_ros_msgs
--- stderr: depthai_ros_msgs                         
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< depthai_ros_msgs [0.10s, exited with code 1]

Summary: 0 packages finished [0.26s]
  1 package failed: depthai_ros_msgs
  1 package had stderr output: depthai_ros_msgs
  1 package not processed
```

Basically the CMakeList.txt is using catkin to compile the packages and I do not have it install. 
I have tried to compile the depthai_bridge package alone using the ament_cmake compilation tool (just uncommented this part and commented the catkin part) and still the same error. It is to be expected since the depthai_ros_msgs only has catkin options and not ament.

I would like to know if it is required to install catkin in ROS2 in order to compile the packages or are there any modifications we can make to compile them using ament?
I would like to be sure about this, before installing catkin in ROS2 which does not look that trivial.

Thank you!

saching13, 2021-05-17:
Don't install catkin. 
You might need to use `foxy-devel` branch. 
However there are some fixes I still need to add to that branch to make it work. 
Will get it up and running by tomorrow and get back to you.

luiscastro1995, 2021-05-17:
Forgot to mention, I am using foxi-devel already.
Ok perfect, let me know as soon as you have something!

saching13, 2021-05-18:
Hey @luiscastro1995 ,
I have added instructions on how-to and everything. Can you try with this in a clean workspace may be ?

And as of examples I haven't written launch files or added all the nodes as executable yet. 
`stereo_node` is working.
Other nodes should work. but not tested yet.

luiscastro1995, 2021-05-20:
Hey @saching13 sorry or the delay.
I have followed your new guide and everything is working fine!
Thanks

closed 2021-06-04 by saching13