luxonis/depthai-ros #14: Dependencies installation failed

ans-qureshi, 2021-06-02:
I get this error after running `sudo wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/noetic-devel/install_dependencis.sh | bash
`

[image]

ans-qureshi, 2021-06-02:
Issue resolved after replacing line 10 of depthai-ros/install_dependencis.sh i.e. `cmake --build build --target install` with `sudo cmake --build build --target install`

@saching13 could you make suitable changes please?

closed 2021-06-02 by ans-qureshi

saching13, 2021-06-02:
`sudo wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/noetic-devel/install_dependencis.sh | bash`

This command is run with `sudo` so any command inside the file will run as root user. So, it is normally not necessary in any ubuntu.

themarpe, 2021-06-03:
@saching13 @ans-qureshi I think that the issue here is that the `sudo` applies to `wget` but not to `bash` command.
One option would be to do: 
```
sudo bash -c 'wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/noetic-devel/install_dependencis.sh | bash'
```
or maybe just 
```
wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/noetic-devel/install_dependencis.sh | sudo bash
```

Other dependencies work as expected as they do `sudo` in their own script for commands that need it.
So could also just drop `sudo` from the installation command:
```
wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/noetic-devel/install_dependencis.sh | bash
``` 
And then do `sudo` in the script itself which need it (namely - installing to `/usr/local`, the aftermentioned `sudo cmake --build build --target install` by @ans-qureshi)

saching13, 2021-06-03:
Hhmm. Makes sense. Will fix it. 
I am not sure why I didn't experience it or why other users didn't report this.