luxonis/depthai-ros #15: Change the point cloud display orientation

yo4hi6o, 2021-06-11:
Hi!

I was able to display the point cloud with rviz with'roslaunch depthai_examples stereo_nodelet.launch'.

The orientation of the point cloud display displayed in rviz is landscape by default, but I would like to change the orientation of the point cloud display from top to bottom. Where can I adjust it?

Thanks!

saching13, 2021-06-11:
it is displayed based on where the map origin is set on rviz. I have set it to the base of the URDF file of OAK-D in the example.

Can I know why you would want to change that and the use case of changing in the visualization?

yo4hi6o, 2021-06-18:
Thank you saching13!
The purpose is to display the position of the picking object by the manipulator as a point cloud by the oak-d camera.
The position of the camera is fixed from above.

saching13, 2021-06-18:
And do you have a TF link from the Manipulator to the OAK-D's state publisher baselink ?

yo4hi6o, 2021-06-22:
> And do you have a TF link from the Manipulator to the OAK-D's state publisher baselink ?

closed 2021-06-22 by yo4hi6o