luxonis/depthai-ros #16: Fatal error when running yolov4_publisher.launch

roboav8r, 2021-06-16:
Hi all,
I am running Ubuntu 20.04/ROS Noetic on a 

When I execute ```roslaunch depthai_examples yolov4_publisher.launch``` I get the following error:

```
...
[14442C10C11690D000] [459.433] [system] [warning] Stereo subpixel/extended mode is not stable when one of the following nodes are enabled: 'NeuralNetwork, ImageManip, SpatialCalculator'
[14442C10C11690D000] [459.434] [StereoDepth(4)] [warning] Median filter usage with LR-check/Extended/Subpixel is not yet implemented. The filter will be disabled
[14442C10C11690D000] [459.483] [SpatialDetectionNetwork(1)] [warning] Network compiled for 6 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
[14442C10C11690D000] [466.814] [system] [critical] Fatal error. Please report to developers. Log: 'StereoSipp' '531'
terminate called after throwing an instance of 'std::runtime_error'
  what():  Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'detections' (X_LINK_ERROR)'
================================================================================REQUIRED process [yolov4_publisher-3] has died!
process has died [pid 39648, exit code -6, cmd /home/jd/catkin_ws/devel/lib/depthai_examples/yolov4_spatial_node __name:=yolov4_publisher __log:=/home/jd/.ros/log/2854799e-ceeb-11eb-b497-b553a812bbd9/yolov4_publisher-3.log].
log file: /home/jd/.ros/log/2854799e-ceeb-11eb-b497-b553a812bbd9/yolov4_publisher-3*.log
Initiating shutdown!
================================================================================
...
```

Has anyone else encountered this and is there a known fix?
Thanks!

Luxonis-Brandon, 2021-06-16:
Yes so currently subpixel and/or extended disparity modes are not stable in combination with NeuralNetwork, ImageManip, and/or SpatialCalculator nodes.

We're trying to figure out what exactly is causing the lockup, but haven't been able to do so yet.

So for now if you disable subpixel and extended disparity this issue should go away.

roboav8r, 2021-06-16:
Thanks Brandon.
I see a couple ```.launch``` files with ```<arg name="extended" default="false" />``` or ```<arg name="subpixel" default="false" />``` but they're commented out. Can I set these directly in ```yolov4_publisher.launch``` or do I need to modify the source code directly?

Thanks again!
--JD

saching13, 2021-06-16:
Hey @roboav8r ,
They are not settable as of now from launch files. I changed the code to disable subpixel for now.

I will be restructuring the examples to make them more configurable from launch files tomorrow.

roboav8r, 2021-06-16:
Nice. That worked and I'm getting the output now. Thanks again @saching13 @Luxonis-Brandon

closed 2021-06-16 by roboav8r