luxonis/depthai-ros #20: Unrecognized image encoding [nv12]

VineetTambe, 2021-08-07:
I am very new to depthai and am trying to interface the OAK1 camera. 

For now, I want to publish the video stream on a particular topic which I was able to do by remapping it in the rgb_publisher .launch in the depthai_example package. However, when I try to view the stream using rqt_image_view  it throws the following error:
`ImageView.callback_image() while trying to convert image from 'nv12' to 'rgb8' an exception was thrown (Unrecognized image encoding [nv12])
`
How do I convert the nv12 image frame into something that rqt will understand?
Any help would be appreciated.

VineetTambe, 2021-08-09:
Can anyone help?

Luxonis-Brandon, 2021-08-09:
Sorry about the delay.  I don't personally know on this one.  I will need @saching13 's help.  Will ask him offline.

saching13, 2021-08-09:
Hey @VineetTambe ,
Sorry for the delay. I missed this one. 

So device sends out the rgb image stream as `NV12` format and to allow the idea of being able to record the rosbag and feed it back to the device I left it to be published as `NV12` format. 

But I am working on the new update to make it to `rgb8` and make it optional and have user input the encoding when want to feedback.

saching13, 2021-08-09:
I Will add an update by tomorrow

Serafadam, 2022-11-07:
Closing due to inactivity, if you think this should remain open, please tag me in a comment here.

closed 2022-11-07 by Serafadam