luxonis/depthai-ros #26: Covariance of spatial detections

hwgat, 2021-08-22:
Hi there,

I am currently using multiple OAK-d cameras on a mobile robot for pedestrian detection. I have used the Spatial Tiny-yolo example as a template and then publish detections on a ros topic using rosbridge.  

The tracking package I am using requires geometry_msgs/PoseWithCovariance messages as an input. Is there a way to calculate the covariance of an OAK-d spatial detection?

Cheers,

saching13, 2021-08-23:
I don't think we have it. Let me cross check it and get back to you.

ninoweg, 2023-05-12:
Has there been an update on this? I am using the internal Object Tracker and I would also be interested in obtaining the covariance of the Tracklets. Stating the documentation the tracker uses Kalman Filters, therefore this information should already be there

saching13, 2023-05-17:
Hello, 
This comes from an internal fw from Mx where it doesn't come with an option to enable the covariance of the tracklets.