luxonis/depthai-ros #33: terminate called after throwing an instance of 'std::runtime_error' error when I run rgb_stereo_node.launch

canberkgurel, 2021-09-30:
I am having some trouble running the `rgb_stereo_node.launch` file with my OAK-D. 

```
$ roslaunch depthai_examples rgb_stereo_node.launch
... logging to /home/eia-lab1/.ros/log/66e7356a-220c-11ec-bced-fc777466ce75/roslaunch-eialab1-AERO-15-X9-8369.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/eia-lab1/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://127.0.0.1:46541/

SUMMARY
========

PARAMETERS
 * /oak_description: <?xml version="1....
 * /rgb_stereo_publisher/camera_name: oak
 * /rgb_stereo_publisher/camera_param_uri: package://depthai...
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rgb_stereo_publisher (depthai_examples/rgb_stereo_node)

auto-starting new master
process[master]: started with pid [8406]
ROS_MASTER_URI=http://127.0.0.1:11311

setting /run_id to 66e7356a-220c-11ec-bced-fc777466ce75
process[rosout-1]: started with pid [8428]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [8431]
process[rgb_stereo_publisher-3]: started with pid [8433]
terminate called after throwing an instance of 'std::runtime_error'
  what():  Couldn't find one of the parameters
================================================================================REQUIRED process [rgb_stereo_publisher-3] has died!
process has died [pid 8433, exit code -6, cmd /home/eia-lab1/catkin_ws/devel/lib/depthai_examples/rgb_stereo_node __name:=rgb_stereo_publisher __log:=/home/eia-lab1/.ros/log/66e7356a-220c-11ec-bced-fc777466ce75/rgb_stereo_publisher-3.log].
log file: /home/eia-lab1/.ros/log/66e7356a-220c-11ec-bced-fc777466ce75/rgb_stereo_publisher-3*.log
Initiating shutdown!
================================================================================
[rgb_stereo_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
```

This is the last commit in the package. Let me know if it needs to be updated. Thanks in advance.

```
commit c1a35f84cdfcb01bb7764c2d4a6f4b63c85acfd4 (HEAD -> noetic-devel, origin/noetic-devel, origin/HEAD)
Merge: 13e8722 4161eec
Author: Sachin Guruswamy <43363595+saching13@users.noreply.github.com>
Date:   Tue Aug 10 16:48:41 2021 -0700

    Merge pull request #9 from luxonis/noetic-examples-update
    
    Noetic examples sync with C++ examples - 1

```

saching13, 2021-09-30:
Issue was fixed yesterday and updated to `noetic-devel`. Can you pull it and test it.

canberkgurel, 2021-09-30:
@saching13 Thanks, that fixed the problem!

One follow-up question. I was expecting this launch file to drive the RGB camera as well as the Stereo-pair. However, it looks like **it's only publishing the RGB and Depth**, but not the Stereo-pair camera images. Is it possible to also publish the left and right stereo images?

```
/rgb_stereo_publisher/color/camera_info
/rgb_stereo_publisher/color/image
/rgb_stereo_publisher/color/image/compressed
/rgb_stereo_publisher/color/image/compressed/parameter_descriptions
/rgb_stereo_publisher/color/image/compressed/parameter_updates
/rgb_stereo_publisher/color/image/compressedDepth
/rgb_stereo_publisher/color/image/compressedDepth/parameter_descriptions
/rgb_stereo_publisher/color/image/compressedDepth/parameter_updates
/rgb_stereo_publisher/color/image/theora
/rgb_stereo_publisher/color/image/theora/parameter_descriptions
/rgb_stereo_publisher/color/image/theora/parameter_updates
/rgb_stereo_publisher/stereo/camera_info
/rgb_stereo_publisher/stereo/depth
/rgb_stereo_publisher/stereo/depth/compressed
/rgb_stereo_publisher/stereo/depth/compressed/parameter_descriptions
/rgb_stereo_publisher/stereo/depth/compressed/parameter_updates
/rgb_stereo_publisher/stereo/depth/compressedDepth
/rgb_stereo_publisher/stereo/depth/compressedDepth/parameter_descriptions
/rgb_stereo_publisher/stereo/depth/compressedDepth/parameter_updates
/rgb_stereo_publisher/stereo/depth/theora
/rgb_stereo_publisher/stereo/depth/theora/parameter_descriptions
/rgb_stereo_publisher/stereo/depth/theora/parameter_updates
```

saching13, 2021-09-30:
It is possible. You need to add XlinkOut and call the publisher callbacks on that.

closed 2021-09-30 by canberkgurel