luxonis/depthai-ros #34: source /opt/ros/<ros-distro>/setup.bash: no such file

Charles-Li26, 2021-09-30:
Hi,
I'm new in ROS, so after I successfully did the first 5 steps. Now, in the step 6, there is no such file address of opt/ros/...

Thank you,
CL

saching13, 2021-09-30:
I think you have not installed ros. You need to install the appropriate Ros version depending on your OS.

Charles-Li26, 2021-10-01:
So for the <ros-distro> , does it mean the version of ROS, like I stalled ROS noetic, so I need to fill neotic at that plave. Am I right?

CL

Charles-Li26, 2021-10-01:
So for the <ros-distro >, does it mean the version of ROS, like I stalled ROS noetic, so I need to fill "neotic" at that place. Am I right?

CL

Charles-Li26, 2021-10-01:
So for the "ros-distro", does it mean the version of ROS, like I stalled ROS noetic, so I need to fill "neotic" at that place. Am I right?

CL

saching13, 2021-10-01:
> So for the "ros-distro", does it mean the version of ROS, like I stalled ROS noetic, so I need to fill "neotic" at that place. Am I right?
> 
> CL

Yes.

Charles-Li26, 2021-10-01:
All right, thank you so much. And now I am working on the catkin_make and it shows same error as "ans-qureshi" had. Hopeful that I can solve this error based your comments. 

CL

saching13, 2021-10-01:
But you shouldn't need to pass the path like in #2  since the depthai is installed to the system

Charles-Li26, 2021-10-01:
[image]
With the "catkin_make" commend, I got the same error. But there is one difference which is the #2 has 4 packages which I only got 3 package, the vision-msgs missed.

Charles-Li26, 2021-10-01:
And I cannot find these two documents that mentioned above which are "depthaiConfig.cmake" and "depthai-config.cmake"

saching13, 2021-10-01:
Looks like install dependencies didn't go well on yours. Can you do it again ? it should be installed in `usr/local`
Link [here](https://github.com/luxonis/depthai-ros#install-dependencies)

Charles-Li26, 2021-10-01:
[image]
In the Install Dependencies section, all the other parts are installed right. Just the fist step has some problem and is that caused the mistake in catkin_make?

saching13, 2021-10-01:
it is not. it is failing. Your docker installation keys are missing. The command that is failing is `sudo apt update` I think. You need to fix those ppa installation keys

Charles-Li26, 2021-10-01:
Thank you for your time. All the procedures execute well. And I am moving to the example execution section now. I'm using Oak-D camera to connect to my computer and when I execute the stereo_node.launch commend. I only able to hear a "click" sound from camera and nothing happened. 
[image]
.

saching13, 2021-10-01:
They are being published on ROS topics. you can run `rqt_image_view` and select the topic of the image you want to see. or use `rviz`

Charles-Li26, 2021-10-01:
Thank you so much for bearing me. The rqt_image_view and rviz are working well now.

closed 2021-10-01 by Charles-Li26