luxonis/depthai-ros #45: Resolution mismatch when running rgb_stereo_node.launch

RaduCorcodel, 2021-11-02:
Hi,
I'm running the rgb_stereo_node.launch and in RViz when I want to visualize the pointcloud, I'm met with this error:

```
Error updating depth cloud: Depth image resolution (1280x720) does not match color image resolution (1920x1080)
```

Is there a way to configure the rgb_stereo_publisher/color/image publisher to published at 1280x720 ?

Also, it would be easier for integration if the node would publish a message of sensor_msgs/PointCloud2. That's what all other cameras are publishing including the Realsense and OpenNI-type. Given that those were very popular for years, most vision ROS nodes (including ours) are developed for that type of messages. Just a suggestion, but I think it's pretty important to have.

saching13, 2021-11-02:
It is being added in this PR with RGB alignment. (Note that it will drop the depth fps since it is not well optimized on device of now)  
[https://github.com/luxonis/depthai-ros-examples/pull/12](url)

saching13, 2021-11-03:
and as of the `sensor_msgs/PointCloud2` agree.
We are publishing `sensor_msgs/PointCloud2`. We use depth image proc package for that conversion as of now. http://wiki.ros.org/depth_image_proc#depth_image_proc.2Fpoint_cloud_xyz

Serafadam, 2022-11-07:
Closing due to inactivity, if you think this should remain open, please tag me in a comment here.

closed 2022-11-07 by Serafadam