luxonis/depthai-ros #47: Convert Depth to Meters internally

saching13, 2021-11-30:
Change depth image which is in millimeters to meters to align with ROS standard REP 103 and encoded as 32-bit float as in REP 118,

### References:
- [REP 118](https://www.ros.org/reps/rep-0118.html
)
- [REP 103](https://www.ros.org/reps/rep-0103.html)

peci1, 2022-01-11:
REP 118 explicitly mentions the OpenNI format which is uint16 milimeters. So technically, the output is according to the REP. The performance considerations mentioned at the end of REP 118 might still hold for some use-cases. So maybe it'd be best to support also the float32 output, but let the user choose using a parameter.

saching13, 2022-01-11:
Agreed. I think thats the way to go. But conversion to float32 mostly will happen on host-side instead of on-device in this case.