luxonis/depthai-ros #48: catkin_make linking error

Armanfidan, 2021-12-07:
Hi!

I have checked the other catkin_make issues, but none of them were similar - it would be amazing if you could help. I'm building depthai-ros inside a Docker container. I have built the dependencies, plus depthai-core. When I try to run catkin_make, everything is built without issues until depthai_bridge, but then I get the following:
```
[ 10%] Built target depthai_ros_msgs_generate_messages_cpp
[ 22%] Built target depthai_ros_msgs_generate_messages_py
[ 32%] Built target depthai_ros_msgs_generate_messages_lisp
[ 42%] Built target depthai_ros_msgs_generate_messages_nodejs
[ 54%] Built target depthai_ros_msgs_generate_messages_eus
[ 54%] Built target depthai_ros_msgs_generate_messages
[ 55%] Linking CXX shared library /catkin_ws/devel/lib/libdepthai_bridge.so
/usr/bin/ld: /catkin_ws/src/luxonis/depthai-core/build/install/lib/libdepthai-core.a(CalibrationHandler.cpp.o): relocation R_X86_64_PC32 against symbol `_ZTISt13runtime_error@@GLIBCXX_3.4' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [luxonis/depthai-ros/depthai_bridge/CMakeFiles/depthai_bridge.dir/build.make:236: /catkin_ws/devel/lib/libdepthai_bridge.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:1646: luxonis/depthai-ros/depthai_bridge/CMakeFiles/depthai_bridge.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4 -l4" failed
```

saching13, 2021-12-07:
Sorry for the trouble. 
Can you share the steps followed here, ROS version and OS here ?

Armanfidan, 2021-12-08:
Hi again,
The Docker container has ROS Noetic and Ubuntu 20.04 Focal.
With rosdep installed and set up, I followed the following steps:
- Installed ```libopencv-dev``` and ```python3-vcstool```
- Tried to execute ```sudo wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/noetic-devel/install_dependencies.sh | sudo bash```, which resulted in the following:
```
SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"
Failed to send reload request: No such file or directory
```
- This was caused by the command ```sudo udevadm control --reload-rules``` (udev doesn't work properly in Docker containers), so I followed the steps outlined by Brandon in https://github.com/luxonis/depthai/issues/295#issuecomment-755434296
- This time I got the error in https://github.com/luxonis/depthai-core/issues/121#issue-869688639 (This occurred when building XLink and was caused by the following lines in ```XLink.cmake```:
```
if(NOT LIBUSB_INCLUDE_DIR OR NOT LIBUSB_LIBRARY)
        message(FATAL_ERROR "libusb is required")
endif()
```
- I tried to fix this by setting the variables ```LIBUSB_LIBRARY``` and ```LIBUSB_INCLUDE_DIR``` in my .bashrc and passing them to cmake when building XLink, but it kept occurring.
- When I ran ```sudo apt-get -y install cmake libusb-1.0-0-dev build-essential``` as outlined in the previous issue thread, this got resolved (But I assume I will run into problems when trying to connect the OAK, because udev is enabled in Docker).
- Now that the dependencies are installed, I kept following the steps on https://github.com/luxonis/depthai-ros. When I ran ```catkin_make```, I bumped into the issue in https://github.com/luxonis/depthai-ros/issues/2, which turned out to be because I had not installed depthai-core.
- When I installed this, I ran the script again and I ran into the issue outlined here. I do not know how to proceed.

themarpe, 2021-12-10:
We've discussed with @saching13 regarding this issue (just a bump)
Main issue seems to be that depthai-core was build as static library instead of shared. If you want to use a static library you should specify the `-D CMAKE_TOOLCHAIN_FILE=$PWD/cmake/toolchain/pic.cmake` to create relocatable code that can be linked into a shared library at the end. (Also clear hunter cache `rm -r ~/.hunter` between, as `-fPIC` doesn't properly create a different cache than non `-fPIC` builds)

Armanfidan, 2021-12-11:
Thank you! I had not realised that I'd built depthai-core as a static library, building it with this option did not work but I built it as a dynamic library and it successfully linked.

closed 2021-12-11 by Armanfidan