luxonis/depthai-ros #53: Timestamp issue

aditdoshi333, 2021-12-26:
Hello, I am using depthai oak 1 with ros noetic and melodic. For my use, I am syncing the RGB camera with other components. But there is a timestamp issue with the depthai ros driver.  Sometimes the timestamp of the current frame is old than the last received frame. If you want to verify I can share the bag file also. 

For example, check the timestamp of the frame no 13 in the attached snapshot.
[image]

Please let me know what I can do. The timestamp is very important for syncing. 
Thank you

themarpe, 2021-12-26:
Hi @aditdoshi333 

This large of a difference suggests an issue on the timesyncing part on device. 

Also as of right now the timestamps aren't monotonic unfortunately. We are yet to implement time skeewing to properly account for differences in host and device clocks.

Will take a look at both and get back to you.

themarpe, 2021-12-26:
@aditdoshi333 can you also run with `DEPTHAI_LEVEL=debug` env variable set and let me know the library version? And are you on USB3 @ 5Gbps or USB2 speeds?

aditdoshi333, 2021-12-26:
@themarpe library version I can tell you tomorrow. Yes, I am using a type c port with type c to type c cable. The strange part is this problem is not everytime it is very random.

themarpe, 2021-12-26:
@aditdoshi333 thanks - I got an theory about resolving this completely. Will keep you updated

aditdoshi333, 2021-12-26:
@themarpe thanks waiting for the update

themarpe, 2021-12-27:
@aditdoshi333 
https://github.com/luxonis/depthai-core/pull/310
Feel free to test this `timesync_improvements` branch. 
Optionally you may also disable subsequnent time syncs, by:
```
// Set numSamples to zero - disables Timesync
device.setTimesync(std::chrono::seconds(60), 0, false);
```

Note however that clocks between host and device might skew apart, so longer runs could see timestamps falling out of sync.

I didn't attempt implementing clock skewing on device yet - I'm running a test on how much the clocks skew in my setup. If change won't be too bad, then I'll leave it as that for the time being. Otherwise I'll revisit skewing the clock on device to be able to have monotonic timestamps while keeping in sync with host.

aditdoshi333, 2021-12-27:
@themarpe thanks for the quick update. I am running depthai ros and I don't see any timesync_improvements branch. Can you please guide me through the setup? 

Do I have to build depthai core and then run the depthai ros? 

Thank you

themarpe, 2021-12-27:
@aditdoshi333 yes - build `depthai-core` on that branch, and you'll be able to use this.
Note: I've tested the skew and its kinda bad - after cca 10h saw quite a significant drift. (1h was around a 1-10ms if I haven't made an error in measurement).
So some sort of sync in time would most likely be needed - but without skewing its hard to guarantee monotonic timestamps.

I'll try to revisit this by the end of the week if time permits (or sooner)

themarpe, 2022-10-07:
To be released in v2.18.0 

Skewing is yet to be implemented (and/or capability for a higher priority channel in communication to transfer timesyncing packets)

closed 2022-10-07 by themarpe

matlabbe, 2022-10-18:
We saw a similar issue on v2.17.4, the stamps are jumping back and forth 4 seconds in future. This happens only the first time I start the camera just after I rebooted the computer. Here is a log example of the left camera_info (oak-d) `header/stamp/secs`:
```
1666115253
---
1666115254
---
1666115254
---
1666115258           <----------------- jump +4 seconds
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115258
---
1666115259
---
1666115254                  <----------------- jump -4 seconds
---
1666115254
---
1666115254
---
1666115254
---
1666115254
---
1666115254
---
1666115255
---
1666115255
---
1666115255
---
1666115259             <----------------- jump +4 seconds
---
1666115259
---
1666115259
---
1666115259
---
1666115259
---
1666115259
---
1666115259
---
1666115259
---
1666115259
---
1666115259
---
1666115259
---
1666115259
---
1666115259
---
1666115259
---
1666115259
---
1666115255                 <----------------- jump -4 seconds
---
1666115255
---
1666115255
---
1666115255
---
1666115255
---
1666115255
---
1666115255
---
1666115256
---
1666115256
---
1666115256
---
1666115260             <----------------- jump +4 seconds
---
1666115260
---
1666115260
---
1666115260
---
1666115260
---
1666115260
---
1666115260
---
1666115260
---
1666115260
---
1666115260
---
1666115260
---
1666115260
---
1666115260
---
1666115260
---
1666115260
---
1666115261
---
1666115256                      <----------------- jump -4 seconds
---
1666115256
---
1666115256
---
1666115256
---
1666115256
---
1666115257
---
1666115257
---
1666115257
---
1666115257
---
1666115261           <----------------- jump -4 seconds
---
1666115261
---
166611526
```

Will try to test on v2.18.0 soon and give feedback!

matlabbe, 2022-10-18:
We just tried [develop](https://github.com/luxonis/depthai-core/commits/develop) branch including this commit: https://github.com/luxonis/depthai-core/pull/310
and our issue (jumping 4 sceonds back and forth) is still happening. Note that this issue happens only the first time we launch the camera after a fresh reboot of the computer. Killing the node then relaunch "fixes" the problem, but that would be nicer that it works the first time when we launch the stack on the robot.

themarpe, 2022-10-18:
@matlabbe 
Thanks for the feedback - do you mind printing output of `dai::Clock::now().time_since_epoch()` alongside just for a test?
Also, double checking, these are USB, OAK-D, devices correct, not PoE?

One extra thing to test is running timesync only once: `device.setTimesync(0, 0, false)` and then check the results.

Based on your observation though, I kinda lean more towards the hosts clock jumping although the underlying clock used (`steady_clock`) should be guaranteed to be monotonic.

Let me know about your findings

matlabbe, 2022-10-18:
I put this warning log inside `getFrameTime` function:
```cpp
DEPTHAI_ROS_WARN_STREAM("PRINT TIMESTAMP: ", 
        "steadyHostClock -> " << std::chrono::duration_cast<std::chrono::duration<double>>(dai::Clock::now().time_since_epoch()).count()  << 
        "  steadyBaseTime -> " << std::chrono::duration_cast<std::chrono::duration<double>>(steadyBaseTime.time_since_epoch()).count() << 
        "  currTimePoint -> " << std::chrono::duration_cast<std::chrono::duration<double>>(currTimePoint.time_since_epoch()).count() << 
        "  rosStamp -> " << rosStamp << 
        "  rosBaseTime -> " << rosBaseTime);
```
over this:
https://github.com/luxonis/depthai-ros/blob/9ac79711a7ae247588a86c75911ddb306e03617f/depthai_bridge/include/depthai_bridge/depthaiUtility.hpp#L111

with:
```cpp
#include "depthai/utility/Clock.hpp"
```

Here is a log from the start after I subscribed to one topic:
```
[ INFO] [1666135673.781303483]: Created pipeline to device: auto
[ INFO] [1666135675.812807079]: Set up pipeline to device: auto
[ WARN] [1666135680.918864318]: steadyHostClock -> 38.3286  steadyBaseTime -> 33.2447  currTimePoint -> 38.0179  rosStamp -> 1666135680.608118131  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135680.958042529]: steadyHostClock -> 38.3678  steadyBaseTime -> 33.2447  currTimePoint -> 38.0513  rosStamp -> 1666135680.641470889  rosBaseTime -> 1666135675.834952576
<<<< jump >>>
[ WARN] [1666135681.004375099]: steadyHostClock -> 38.4141  steadyBaseTime -> 33.2447  currTimePoint -> 42.4129  rosStamp -> 1666135685.003139874  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.039231147]: steadyHostClock -> 38.449  steadyBaseTime -> 33.2447  currTimePoint -> 42.4463  rosStamp -> 1666135685.036491741  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.085391665]: steadyHostClock -> 38.4952  steadyBaseTime -> 33.2447  currTimePoint -> 42.4796  rosStamp -> 1666135685.069842879  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.119305109]: steadyHostClock -> 38.5291  steadyBaseTime -> 33.2447  currTimePoint -> 42.513  rosStamp -> 1666135685.103194595  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.158675904]: steadyHostClock -> 38.5684  steadyBaseTime -> 33.2447  currTimePoint -> 42.5463  rosStamp -> 1666135685.136546597  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.204331040]: steadyHostClock -> 38.6141  steadyBaseTime -> 33.2447  currTimePoint -> 42.613  rosStamp -> 1666135685.203248915  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.239587121]: steadyHostClock -> 38.6493  steadyBaseTime -> 33.2447  currTimePoint -> 42.6464  rosStamp -> 1666135685.236601479  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.285734344]: steadyHostClock -> 38.6955  steadyBaseTime -> 33.2447  currTimePoint -> 42.6797  rosStamp -> 1666135685.269952012  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.319184156]: steadyHostClock -> 38.729  steadyBaseTime -> 33.2447  currTimePoint -> 42.7131  rosStamp -> 1666135685.303303515  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.358637699]: steadyHostClock -> 38.7684  steadyBaseTime -> 33.2447  currTimePoint -> 42.7464  rosStamp -> 1666135685.336655455  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.404594180]: steadyHostClock -> 38.8143  steadyBaseTime -> 33.2447  currTimePoint -> 42.8132  rosStamp -> 1666135685.403358004  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.439644795]: steadyHostClock -> 38.8494  steadyBaseTime -> 33.2447  currTimePoint -> 42.8465  rosStamp -> 1666135685.436709362  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.486115004]: steadyHostClock -> 38.8959  steadyBaseTime -> 33.2447  currTimePoint -> 42.8799  rosStamp -> 1666135685.470060945  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.519469854]: steadyHostClock -> 38.9292  steadyBaseTime -> 33.2447  currTimePoint -> 42.9132  rosStamp -> 1666135685.503412569  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.558441688]: steadyHostClock -> 38.9682  steadyBaseTime -> 33.2447  currTimePoint -> 42.9466  rosStamp -> 1666135685.536764414  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.604846950]: steadyHostClock -> 39.0146  steadyBaseTime -> 33.2447  currTimePoint -> 43.0133  rosStamp -> 1666135685.603467111  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.639875655]: steadyHostClock -> 39.0496  steadyBaseTime -> 33.2447  currTimePoint -> 43.0466  rosStamp -> 1666135685.636818487  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.686814985]: steadyHostClock -> 39.0966  steadyBaseTime -> 33.2447  currTimePoint -> 43.08  rosStamp -> 1666135685.670169888  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.719459243]: steadyHostClock -> 39.1292  steadyBaseTime -> 33.2447  currTimePoint -> 43.1133  rosStamp -> 1666135685.703521581  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.759116905]: steadyHostClock -> 39.1689  steadyBaseTime -> 33.2447  currTimePoint -> 43.1467  rosStamp -> 1666135685.736873122  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.804961316]: steadyHostClock -> 39.2147  steadyBaseTime -> 33.2447  currTimePoint -> 43.2134  rosStamp -> 1666135685.803576747  rosBaseTime -> 1666135675.834952576
[ WARN] [1666135681.839928262]: steadyHostClock -> 39.2497  steadyBaseTime -> 33.2447  currTimePoint -> 43.2467  rosStamp -> 1666135685.836927757  rosBaseTime -> 1666135675.834952576
[... cut ...]
Then with `device_->setTimesync(std::chrono::milliseconds(0), 0, false);`, I have this error after ~7 seconds:
```
terminate called after throwing an instance of 'dai::XLinkWriteError'
  what():  Couldn't write data to stream: '__rpc_main' (X_LINK_ERROR)
================================================================================REQUIRED process [oak_camera_manager-1] has died!
process has died [pid 998, exit code -6, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=oak_camera_manager 
Initiating shutdown!
================================================================================
```

matlabbe, 2022-10-19:
Reproduced on my laptop and another OAK-D camera (USB 3) with latest develop branch of depthai-core and latest version of depthai-ros (on Ubuntu 20.04, ROS Noetic). Steps:
1. Reboot computer
2. `roslaunch depthai_examples rgb_stereo_node.launch`
3. `rostopic echo /rgb_stereo_publisher/color/camera_info/header`

Watch the timestamps jumping.

themarpe, 2022-10-28:
@matlabbe 
I've updated `develop` and added new API

Do you mind testing with `device.setTimesync(false)` to disable timesync service (after the initial syncing already happens once)

Let me know how timestamps look in that case

matlabbe, 2022-10-31:
Just tested with `develop` (27ba35985f3f72f062f0ecb363b7ecf58e18678c) and I don't see the problem anymore. I rebooted multiple times and it was fine. Tried `main` gain after just to make sure it is not a computer difference (using a different computer to test) and the problem was there, thus `develop` seems to have fixed the issue!

themarpe, 2022-10-31:
Thanks for reporting back @matlabbe! - we've also modified another thing (onboard device time is now reset to 0 upon start), so that might've be the root cause.