luxonis/depthai-ros #55: colcon build gets stuck on depthai_examples at 71%

JHPHELAN, 2021-12-29:
`ubuntu@AUDACITY:~/depthai-ros_ws$ colcon build --packages-select depthai_examples

Starting >>> depthai_examples
[Processing: depthai_examples]
[Processing: depthai_examples]pthai_examples:build 71% - 1min 2.5s]
[1min 32.3s] [0/1 complete] [depthai_examples:build 71% - 1min 31.7s]`
seems to want to get stuck here
The other 2, depthai_bridge, depthai_ros_msg build ok
ROS2 / Ubuntu 20.04 / Raspberry Pi 4b 4GB

saching13, 2021-12-29:
It has run out of memory to compile. 
`MAKEFLAGS="-j 1" colcon build --packages-select depthai_ros_msgs depthai_bridge` do this to avoid parallelization which will be able to build on pi without issues.

JHPHELAN, 2021-12-29:
@saching13 
Thanks for the tip!  I suspected the same but you showed me the way out.
`ubuntu@AUDACITY:~$ MAKEFLAGS=" -J 1" colcon build --packages-select depthai_examples`
by itself got stuck again.  But I created an 8GB swapfile and ran it again.  After [4min 47s] Success!
Thanks!

closed 2021-12-29 by JHPHELAN