luxonis/depthai-ros #56: IMU timestamp is not reflected in ROS message

nindanaoto, 2022-01-06:
Dear maintainers,

As discussed in the ros channel in discord, the current ImuConverter implementation does not reflect the timestamp of IMU data and only publishes the latest data in a batch. It is preferable that to correct the timestamp in the IMU message by the timestamp of the device like ImageConverter does.

saching13, 2022-01-06:
Idea is to update the BridgePublisher to take in batches and publish the batches quickly in the next release
And as of the timestamp. Device sends unique timestamp for accelerometer, gyro, and rotation vector. Followed by each of them published at different freq. So the idea is to use this info and add the interpolation for the lower freq ones or drop the higher freq packets to match the same freq since ROS message carries all of these in the same IMU message

nindanaoto, 2022-01-09:
I found that the timestamps of Images are synchronized to the host. They have two timestamps for each, ts and tsDevice.
https://github.com/luxonis/depthai-shared/blob/510721a6001be4081340737134dfe1fe3c189d62/include/depthai-shared/datatype/RawImgFrame.hpp#L171-L172
However, IMU has only one timestamp. I guess that this is the raw device timestamp, not synchronized with the host.
https://github.com/luxonis/depthai-shared/blob/510721a6001be4081340737134dfe1fe3c189d62/include/depthai-shared/datatype/RawIMUData.hpp#L27
Therefore, I think the best solution for this issue is modifying the firmware and depthai-share to support the host synchronized timestamp for IMU.

Luxonis-Brandon, 2022-01-09:
Thanks.  Asking internally.

szabi-luxonis, 2022-01-09:
> I found that the timestamps of Images are synchronized to the host. They have two timestamps for each, ts and tsDevice. [...]

IMU timestamps you mentioned are synchronized with host.

nindanaoto, 2022-01-18:
I've just tried to fix the timestamp of the ROS message by the way like ImageConverter.cpp is doing.
```
// setting the header
    outImuMsg.header.seq = _sequenceNum;
    const auto imuPacket = inData->packets[inData->packets.size() - 1];
#ifndef IS_ROS2
    auto tstamp = imuPacket.gyroscope.timestamp.get();
    auto rosNow = ::ros::Time::now();
    auto steadyTime = std::chrono::steady_clock::now();
    auto diffTime = steadyTime - tstamp;
    long int nsec = rosNow.toNSec() - diffTime.count();
    auto rosStamp = rosNow.fromNSec(nsec);
    outImuMsg.header.stamp = rosStamp;
#else
    outImuMsg.header.stamp = rclcpp::Clock().now();
#endif
```
However, this causes this error message in kalibr.
```
[FATAL] [1642519236.833727]: The time ranges of the camera and IMU do not overlap. Please make sure that your sensors are synchronized correctly.
```
Are the timestamp of the IMU and the one of the stereo camera images using same time sources?

themarpe, 2022-01-18:
CC: @saching13

Should be the same timesource, but the difference is way to big. What if you compare the "raw" timestamps before ROS conversion?

nindanaoto, 2022-01-19:
The final output of this code gives `0:00:00.018791`. This is bigger than expected (1ms order) but much smaller than the difference after ROS conversion.  
```
#!/usr/bin/env python3

import cv2
import depthai as dai
import datetime

# Create pipeline
pipeline = dai.Pipeline()

# Define sources and outputs
monoLeft = pipeline.create(dai.node.MonoCamera)
monoRight = pipeline.create(dai.node.MonoCamera)
xoutLeft = pipeline.create(dai.node.XLinkOut)
xoutRight = pipeline.create(dai.node.XLinkOut)
xlinkOut = pipeline.create(dai.node.XLinkOut)
imu = pipeline.create(dai.node.IMU)

xoutLeft.setStreamName('left')
xoutRight.setStreamName('right')
xlinkOut.setStreamName("imu")

# Properties
monoLeft.setBoardSocket(dai.CameraBoardSocket.LEFT)
monoLeft.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)
monoRight.setBoardSocket(dai.CameraBoardSocket.RIGHT)
monoRight.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)

# Linking
monoRight.out.link(xoutRight.input)
monoLeft.out.link(xoutLeft.input)

# enable ACCELEROMETER_RAW and GYROSCOPE_RAW at 500 hz rate
imu.enableIMUSensor([dai.IMUSensor.ACCELEROMETER_RAW, dai.IMUSensor.GYROSCOPE_RAW], 500)
# above this threshold packets will be sent in batch of X, if the host is not blocked and USB bandwidth is available
imu.setBatchReportThreshold(1)
# maximum number of IMU packets in a batch, if it's reached device will block sending until host can receive it
# if lower or equal to batchReportThreshold then the sending is always blocking on device
# useful to reduce device's CPU load  and number of lost packets, if CPU load is high on device side due to multiple nodes
imu.setMaxBatchReports(10)

# Link plugins IMU -> XLINK
imu.out.link(xlinkOut.input)

diff = []
# Connect to device and start pipeline
with dai.Device(pipeline) as device:

    # Output queue for imu bulk packets
    imuQueue = device.getOutputQueue(name="imu", maxSize=50, blocking=False)

    # Output queues will be used to get the grayscale frames from the outputs defined above
    qLeft = device.getOutputQueue(name="left", maxSize=4, blocking=False)
    qRight = device.getOutputQueue(name="right", maxSize=4, blocking=False)
    i = 0
    while i < 100:
        imuData = imuQueue.get()  # blocking call, will wait until a new data has arrived
        inLeft = qLeft.tryGet()
        if inLeft is not None:
            print(i)

            imuPackets = imuData.packets
            mindiff = datetime.timedelta(days=1)
            for imuPacket in imuPackets:
                gyroValues = imuPacket.gyroscope
                gyroTs = gyroValues.timestamp.get()
                mindiff = min(mindiff,abs(gyroTs-inLeft.getTimestamp()))
            print(gyroTs)
            print(inLeft.getTimestamp())
            diff.append(mindiff)
            i += 1

print(sum(diff,datetime.timedelta(0))/100)
```

nindanaoto, 2022-01-19:
Retrying to check the difference after ROS conversion gives much smaller (less than 1 second) values...... I'm guessing that some time drift is happening.

saching13, 2022-01-19:
I will check this issue by next week.

nindanaoto, 2022-02-04:
I've confirmed that we can publish raw accelerometer and gyro data at 500Hz by using batches.
Here is the test code.
https://github.com/nindanaoto/oakd-ros-simple

saching13, 2022-11-09:
Fixed in adding Interpolation for IMU

closed 2022-11-09 by saching13