luxonis/depthai-ros #57: install_dependencies.sh fails if uvcdynctrl not installed

NovoG93, 2022-01-07:
Hello there,

the [install_dependencies.sh](https://github.com/luxonis/depthai-ros/blob/main/install_dependencies.sh#L2) from http://docs.luxonis.com/_static/install_dependencies.sh fails on line 127 `UVC_DYN_INSTALLED=$(dpkg-query -W --showformat='${Status}\n' uvcdynctrl|grep "install ok installed")` if uvcdynctrl is not installed on the system.

`UVC_DYN_INSTALLED=$(dpkg-query -W --showformat='${Status}\n' uvcdynctrl|grep "install ok installed")` returns 1 if it is not installed and with `set -e` in line 3 the code exists here.

saching13, 2022-01-07:
it's an assignment to variable. it shouldn't exit for that. Are you using bash shell ? and can I know the system specs ?

roni-kreinin, 2022-01-10:
I am getting the same issue on an RPI4B 4GB.If I install uvcdynctrl then I get this error:
```
CMake Error at cmake/XLink.cmake:26 (message):
  libusb is required
Call Stack (most recent call first):
  CMakeLists.txt:8 (include)

-- Configuring incomplete, errors occurred!
See also "/root/.hunter/_Base/9b4e732/f61e57a/90457df/Build/XLink/Build/XLink-Release-prefix/src/XLink-Release-build/CMakeFiles/CMakeOutput.log".
See also "/root/.hunter/_Base/9b4e732/f61e57a/90457df/Build/XLink/Build/XLink-Release-prefix/src/XLink-Release-build/CMakeFiles/CMakeError.log".
make[2]: *** [CMakeFiles/XLink-Release.dir/build.make:111: XLink-Release-prefix/src/XLink-Release-stamp/XLink-Release-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/XLink-Release.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

[hunter ** FATAL ERROR **] Build step failed (dir: /root/.hunter/_Base/9b4e732/f61e57a/90457df/Build/XLink
[hunter ** FATAL ERROR **] [Directory:/tmp/depthai-core/cmake]

------------------------------ ERROR -----------------------------
    https://hunter.readthedocs.io/en/latest/reference/errors/error.external.build.failed.html
------------------------------------------------------------------

CMake Error at /root/.hunter/_Base/Download/Hunter/0.23.320/9b4e732/Unpacked/cmake/modules/hunter_error_page.cmake:12 (message):
Call Stack (most recent call first):
  /root/.hunter/_Base/Download/Hunter/0.23.320/9b4e732/Unpacked/cmake/modules/hunter_fatal_error.cmake:20 (hunter_error_page)
  /root/.hunter/_Base/Download/Hunter/0.23.320/9b4e732/Unpacked/cmake/modules/hunter_download.cmake:623 (hunter_fatal_error)
  /root/.hunter/_Base/Download/Hunter/0.23.320/9b4e732/Unpacked/cmake/modules/hunter_add_package.cmake:53 (hunter_download)
  cmake/depthaiDependencies.cmake:9 (hunter_add_package)
  CMakeLists.txt:134 (include)

-- Configuring incomplete, errors occurred!
See also "/tmp/depthai-core/build/CMakeFiles/CMakeOutput.log".
```

saching13, 2022-01-10:
@themarpe  any idea on this ?

themarpe, 2022-01-11:
@roni-kreinin regarding your issue - can you install libusb dev library using: `sudo apt install libusb-1.0-0-dev`
Does that resolve the issue for you?

@saching13 can you try reproducing the OP issue? If it shows up, would be a blocker in those cases.

roni-kreinin, 2022-01-11:
@themarpe Yes this fixes the issue. I do still get the warning that I shouldn't have uvcdynctrl installed however.

themarpe, 2022-01-11:
@roni-kreinin great
We are fixing the install_dependencies.sh script, which should address the rest of this issue:)

NovoG93, 2022-01-21:
> ou try reproducing the

Yes I aam using bash shell.

OS: Ubuntu 20.04

themarpe, 2022-01-21:
@NovoG93 the script should be updated now (https://github.com/luxonis/depthai-python/pull/483)
Let us know if it works for you

Serafadam, 2022-11-07:
Closing due to inactivity, if you think this should remain open, please tag me in a comment here.

closed 2022-11-07 by Serafadam