luxonis/depthai-ros #59: Build as per instructions fail on Raspberry Pi Ubuntu 20.04

JesseCake, 2022-01-18:
I'm attempting to install this library for ROS on Ubuntu 20.04 on Raspberry Pi 4.

I'm unsure whether this is compatible, however ideally I'd be able to use this as a sensor for my robot requiring a small form factor computer.

Here is the full output:

`ubuntu@ubuntu:~/oakdlite$ catkin_make
Base path: /home/ubuntu/oakdlite
Source space: /home/ubuntu/oakdlite/src
Build space: /home/ubuntu/oakdlite/build
Devel space: /home/ubuntu/oakdlite/devel
Install space: /home/ubuntu/oakdlite/install
Creating symlink "/home/ubuntu/oakdlite/src/CMakeLists.txt" pointing to "/opt/ros/noetic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/ubuntu/oakdlite/src -DCATKIN_DEVEL_PREFIX=/home/ubuntu/oakdlite/devel -DCMAKE_INSTALL_PREFIX=/home/ubuntu/oakdlite/install -G Unix Makefiles" in "/home/ubuntu/oakdlite/build"
####
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/ubuntu/oakdlite/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ubuntu/oakdlite/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - depthai_ros_msgs
-- ~~  - depthai_bridge
-- ~~  - depthai_examples
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'depthai_ros_msgs'
-- ==> add_subdirectory(luxonis/depthai-ros/depthai_ros_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- depthai_ros_msgs: 5 messages, 2 services
-- +++ processing catkin package: 'depthai_bridge'
-- ==> add_subdirectory(luxonis/depthai-ros/depthai_bridge)
CMake Error at luxonis/depthai-ros/depthai_bridge/CMakeLists.txt:24 (find_package):
  Could not find a package configuration file provided by "depthai" with any
  of the following names:

    depthaiConfig.cmake
    depthai-config.cmake

  Add the installation prefix of "depthai" to CMAKE_PREFIX_PATH or set
  "depthai_DIR" to a directory containing one of the above files.  If
  "depthai" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/oakdlite/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/oakdlite/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
`

saching13, 2022-01-18:
Hello @JesseCake ,
Sorry for the issue. Were you able to run `install_dependencies.sh` without any issues before `catkin_make` ?

JesseCake, 2022-01-18:
No problem at all and thanks for being so quick!

Yep no problems at all running install_dependencies.sh

saching13, 2022-01-18:
@JesseCake  looks like the script was stopping after executing the first command. I updated the dependencies script. can you run it agian? It should fix the issues.

JesseCake, 2022-01-18:
Hmmm looks to be the same? (result below)

ubuntu@ubuntu:~/oakdlite$ catkin_make
Base path: /home/ubuntu/oakdlite
Source space: /home/ubuntu/oakdlite/src
Build space: /home/ubuntu/oakdlite/build
Devel space: /home/ubuntu/oakdlite/devel
Install space: /home/ubuntu/oakdlite/install
Creating symlink "/home/ubuntu/oakdlite/src/CMakeLists.txt" pointing to "/opt/ros/noetic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/ubuntu/oakdlite/src -DCATKIN_DEVEL_PREFIX=/home/ubuntu/oakdlite/devel -DCMAKE_INSTALL_PREFIX=/home/ubuntu/oakdlite/install -G Unix Makefiles" in "/home/ubuntu/oakdlite/build"
####
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/ubuntu/oakdlite/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ubuntu/oakdlite/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - depthai_ros_msgs
-- ~~  - depthai_bridge
-- ~~  - depthai_examples
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'depthai_ros_msgs'
-- ==> add_subdirectory(luxonis/depthai-ros/depthai_ros_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- depthai_ros_msgs: 5 messages, 2 services
-- +++ processing catkin package: 'depthai_bridge'
-- ==> add_subdirectory(luxonis/depthai-ros/depthai_bridge)
CMake Error at luxonis/depthai-ros/depthai_bridge/CMakeLists.txt:24 (find_package):
  Could not find a package configuration file provided by "depthai" with any
  of the following names:

    depthaiConfig.cmake
    depthai-config.cmake

  Add the installation prefix of "depthai" to CMAKE_PREFIX_PATH or set
  "depthai_DIR" to a directory containing one of the above files.  If
  "depthai" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/oakdlite/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/oakdlite/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

JesseCake, 2022-01-18:
Here is also the output of the error log:

ubuntu@ubuntu:~/oakdlite$ cat /home/ubuntu/oakdlite/build/CMakeFiles/CMakeError.log 
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/ubuntu/oakdlite/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_8054c/fast && /usr/bin/make -f CMakeFiles/cmTC_8054c.dir/build.make CMakeFiles/cmTC_8054c.dir/build
make[1]: Entering directory '/home/ubuntu/oakdlite/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_8054c.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_8054c.dir/src.c.o   -c /home/ubuntu/oakdlite/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_8054c
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_8054c.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD    -rdynamic CMakeFiles/cmTC_8054c.dir/src.c.o  -o cmTC_8054c 
/usr/bin/ld: CMakeFiles/cmTC_8054c.dir/src.c.o: in function `main':
src.c:(.text+0x48): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x50): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x5c): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_8054c.dir/build.make:87: cmTC_8054c] Error 1
make[1]: Leaving directory '/home/ubuntu/oakdlite/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_8054c/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/ubuntu/oakdlite/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_17ab3/fast && /usr/bin/make -f CMakeFiles/cmTC_17ab3.dir/build.make CMakeFiles/cmTC_17ab3.dir/build
make[1]: Entering directory '/home/ubuntu/oakdlite/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_17ab3.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_17ab3.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_17ab3
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_17ab3.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_17ab3.dir/CheckFunctionExists.c.o  -o cmTC_17ab3  -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_17ab3.dir/build.make:87: cmTC_17ab3] Error 1
make[1]: Leaving directory '/home/ubuntu/oakdlite/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_17ab3/fast] Error 2

saching13, 2022-01-18:
you need to run the following command again. 
`sudo wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/main/install_dependencies.sh | sudo bash`
Can you send me the output of the above command. 

From the first reply it looks like depthai-core is still not installed.

JesseCake, 2022-01-18:
Hmmmm ok so now this script seems to be building vast amounts of stuff. Before it appeared to only pull from apt sources. Is this building everything it requires and then overwriting system apt installed packages?

I'll paste the output once it's finished building

JesseCake, 2022-01-19:
Ok, as requested here's the mega-dump output from running the script. (it ran successfully):

ubuntu@ubuntu:~$ sudo wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/main/install_dependencies.sh | sudo bash
Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:2 http://packages.ros.org/ros/ubuntu focal InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'python-dev-is-python2' instead of 'python-dev'
cmake is already the newest version (3.16.3-1ubuntu1).
libjpeg-dev is already the newest version (8c-2ubuntu8).
libpng-dev is already the newest version (1.6.37-2).
libsm6 is already the newest version (2:1.2.3-1).
libusb-1.0-0-dev is already the newest version (2:1.0.23-2build1).
libxext6 is already the newest version (2:1.3.4-0ubuntu1).
pkg-config is already the newest version (0.29.1-0ubuntu4).
python3 is already the newest version (3.8.2-0ubuntu2).
python3-numpy is already the newest version (1:1.17.4-5ubuntu3).
libdc1394-22-dev is already the newest version (2.2.5-2.1).
libgtk2.0-dev is already the newest version (2.24.32-4ubuntu4).
libtbb-dev is already the newest version (2020.1-2).
libtbb2 is already the newest version (2020.1-2).
python-dev-is-python2 is already the newest version (2.7.17-4).
build-essential is already the newest version (12.8ubuntu1.1).
git is already the newest version (1:2.25.1-1ubuntu3.2).
libgl1-mesa-glx is already the newest version (21.0.3-0ubuntu0.3~20.04.5).
libtiff-dev is already the newest version (4.1.0+git191117-2ubuntu0.20.04.2).
udev is already the newest version (245.4-4ubuntu3.15).
ffmpeg is already the newest version (7:4.2.4-1ubuntu0.1).
libavcodec-dev is already the newest version (7:4.2.4-1ubuntu0.1).
libavformat-dev is already the newest version (7:4.2.4-1ubuntu0.1).
libswscale-dev is already the newest version (7:4.2.4-1ubuntu0.1).
python3-pip is already the newest version (20.0.2-5ubuntu1.6).
The following packages were automatically installed and are no longer required:
  linux-headers-5.4.0-1028-raspi linux-headers-5.4.0-1038-raspi linux-image-5.4.0-1028-raspi linux-image-5.4.0-1038-raspi linux-modules-5.4.0-1028-raspi linux-modules-5.4.0-1038-raspi
  linux-raspi-headers-5.4.0-1028 linux-raspi-headers-5.4.0-1038
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Requirement already satisfied: pip in /usr/local/lib/python3.8/dist-packages (21.3.1)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"
Cloning into 'depthai-core'...
remote: Enumerating objects: 17360, done.
remote: Counting objects: 100% (1084/1084), done.
remote: Compressing objects: 100% (588/588), done.
remote: Total 17360 (delta 717), reused 676 (delta 436), pack-reused 16276
Receiving objects: 100% (17360/17360), 2.88 MiB | 2.51 MiB/s, done.
Resolving deltas: 100% (12741/12741), done.
Submodule 'shared/depthai-bootloader-shared' (https://github.com/luxonis/depthai-bootloader-shared.git) registered for path 'shared/depthai-bootloader-shared'
Submodule 'shared/depthai-shared' (https://github.com/luxonis/depthai-shared.git) registered for path 'shared/depthai-shared'
Cloning into '/tmp/depthai-core/shared/depthai-bootloader-shared'...
remote: Enumerating objects: 228, done.        
remote: Counting objects: 100% (228/228), done.        
remote: Compressing objects: 100% (152/152), done.        
remote: Total 228 (delta 83), reused 210 (delta 68), pack-reused 0        
Receiving objects: 100% (228/228), 32.53 KiB | 708.00 KiB/s, done.
Resolving deltas: 100% (83/83), done.
Cloning into '/tmp/depthai-core/shared/depthai-shared'...
remote: Enumerating objects: 4190, done.        
remote: Counting objects: 100% (1844/1844), done.        
remote: Compressing objects: 100% (646/646), done.        
remote: Total 4190 (delta 1103), reused 1580 (delta 906), pack-reused 2346        
Receiving objects: 100% (4190/4190), 590.63 KiB | 1.95 MiB/s, done.
Resolving deltas: 100% (2497/2497), done.
Submodule path 'shared/depthai-bootloader-shared': checked out '126f344155346836a6a2d67b0993c161b6b48729'
Submodule path 'shared/depthai-shared': checked out '510721a6001be4081340737134dfe1fe3c189d62'
[... 2409 lines cut ...]
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/lzma/index.h
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/lzma/index_hash.h
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/lzma/container.h
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/lzma/delta.h
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/lzma/filter.h
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/lzma/version.h
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/lzma/bcj.h
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/lzma/stream_flags.h
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/lzma/base.h
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/backward.hpp
[... cut ...]
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/nlohmann
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/nlohmann/json.hpp
-- Installing: /usr/local/lib/cmake/depthai/dependencies/include/lzma.h
-- Installing: /usr/local/lib/cmake/depthai/depthaiTargets.cmake
-- Installing: /usr/local/lib/cmake/depthai/depthaiTargets-debug.cmake
-- Installing: /usr/local/lib/cmake/depthai/depthaiConfig.cmake
-- Installing: /usr/local/lib/cmake/depthai/depthaiConfigVersion.cmake
-- Installing: /usr/local/lib/cmake/depthai/depthaiDependencies.cmake

JesseCake, 2022-01-19:
ooooooh boy that took a while. After a few failed attempts at multi-processor building on a 2GB ram Raspberry Pi 4, I ended up backing right down to just one process (of course with -j 1 flag after catkin_make) and also had to create a swap file for ubuntu so I could squeeze every last drop of ram out to successfully build.

But I got there in the end!

Currently just getting a few other things set up on another desktop nearby so I can see the visual output from the stereo example to make sure it was successful.

Thanks so much for your help

saching13, 2022-01-19:
Sorry for the issue. We will release a deb package to install these directly in the fututre.

closed 2022-01-19 by saching13