luxonis/depthai-ros #60: OAK-D Lite troubles with camera parameters mismatch

JesseCake, 2022-01-19:
When I roslaunch the example stereo_node.launch file (and probably the others) I'm met with this error, though the node successfully runs:

```
[18443010A1460B1300] [18.901] [MonoCamera(1)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
[18443010A1460B1300] [18.902] [MonoCamera(0)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
```

When I connect to rosmaster via another machine running rviz, I can open each of the cameras, though they're stretched (meaning there's a resolution mismatch going on), but as soon as I open the stereo depth path I get an error and no point cloud:

`Error updating depth cloud: Depth image size and camera info don't match: 640 x 480 vs 1280 x 720(binning: 1 x 1, ROI size: 1280 x 720)`

I've checked the launch files and also the companion python files, and I can see that there is a camera model field that suggests that I should configure it to be "OAK-D-LITE" but this also doesn't help.

I assume this has to do with camera parameters under the params folder, but im at a loss as to what should be modified here, or perhaps I need to run a calibration routine to sort this out? 

Do you have suitable launch examples and camera parameters for the OAK-D Lite that could maybe be included here?

Luxonis-Brandon, 2022-01-19:
Thanks and sorry about the trouble.  I'll need @saching13 's help here.

JesseCake, 2022-01-19:
Gosh no need to be sorry! Very thankful for your hard work on this, happy to go through the process.

saching13, 2022-01-19:
Hi, 
The example is not updated to contain the `camera_info` I will do it today. Sorry for the issue.

saching13, 2022-01-19:
`roslaunch depthai_examples yolov4_publisher.launch camera_model:=OAK-D-LITE` Currently this example is updated for OAK-D-LITE.

saching13, 2022-01-19:
`roslaunch depthai_examples stereo_node.launch monoResolution:=400p camera_model:=OAK-D-LITE` updated the stereo node for now. LMK if this helps. Thanks.

JesseCake, 2022-02-15:
Excellent - the mono stereo node works well - I don't need colour right now, but in future it might be good. Thanks for all the hard work.

saching13, 2022-02-15:
Awesome. @daxoft  recently updated all the examples in a branch. will be merging in a day or so.  :)

JesseCake, 2022-02-15:
You guys are on fire! Definitely keen on starting to use OAK-D Lites on lots of my works if you're this active in driver development. Definitely keen on a wider angle lens version at some point too (unsure if that's a thing yet?)

saching13, 2022-02-15:
Thanks for the kind words. Let us know if you face any other issues we would be happy to help here or in our Discord Community. 

And Yes. We are working on Wide angle products. They are currently available on our Beta store [here](https://shop.luxonis.com/collections/beta-store)

JesseCake, 2022-02-16:
Ah brilliant! I had a quick look on your website to try and join your Discord to talk directly, but it seems the link is invalid?

saching13, 2022-02-16:
Oh my bad. Here is the link - [https://discord.gg/ju2Qmq8v](https://discord.gg/ju2Qmq8v)

cc: @Erol444  on the invalid link.

Erol444, 2022-02-16:
@JesseCake  where was the link located that was invalid for you? Looking at the website and documentation, we use https://discord.com/invite/4hGT3AFPMZ , which works for me.
Thanks, Erik

JesseCake, 2022-02-17:
@Erol444 it was just on your luxonis website - I think from memory there was a link provided at the bottom of the page or on the contact page? When I clicked it, it simply said the invite was invalid.

saching13, 2022-02-24:
I updated the `stereo_inertial_node` which contains RGB resolution with the right resolution and camera info. 
 In this example, if the stereo is using 720p/800p. RGB camera will publish at 720p and if it is less than that it will publish at 360p.
This is done so that you can get colorized pointCloud output.

And I will be deprecating  the rgb_stereo_node.cpp since everything it does is done by `stereo_intertial_node`

Thoguhts ?

closed 2022-11-09 by saching13