luxonis/depthai-ros #63: Oak d poe with ROS

antoineTK, 2022-01-26:
Hi guys!

I'm not sure this is the right place to be asking, but what is the best way to interface a Oak d poe camera with ROS nowadays? Would I use depthai-ros?

Thanks a lot

Luxonis-Brandon, 2022-01-26:
@saching13 is best to advise.  I think so, but I'm not sure if there are any specifics that need to be taken into account for PoE.

saching13, 2022-01-26:
Yes. you can use depthai-ros. 
For ex: `roslaunch depthai_examples stereo_node.launch monoResolution:=400p camera_model:=OAK-D-POE` from [here](https://github.com/luxonis/depthai-ros-examples/blob/main/depthai_examples/launch/stereo_node.launch)

antoineTK, 2022-01-27:
sounds good! thank you guys

I have another question, regarding the poe injector. I see it needs to be 802.3af compliant, what about the speed? Would a 100MBps injector suffice?

themarpe, 2022-01-27:
@antoineTK 
1Gbps is highly recommended, as even at those speeds its much slower than USB models at USB3 Gen1 speeds

antoineTK, 2022-01-27:
Got it thank you! I guess the 4k video will take most of that bandwidth, could you recommend a poe injector?

themarpe, 2022-01-27:
@antoineTK 
More info on powering PoE devices in general here: https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/powering_poe_devices.html

And WRT to injector we usually recommend: https://www.amazon.com/gp/product/B07JCB5XWF/

antoineTK, 2022-01-28:
oh i missed that, great info on poe devices!

@saching13 to save on bandwidth can RGB image stream be disabled altogether? I see in the launch file I can change the resolution. We would only work with depth at the moment

saching13, 2022-01-30:
I will modify the example to make that an argument.

closed 2022-11-09 by Serafadam

RhythmDJay, 2022-12-08:
@Serafadam Im currently trying to run depthai_examples/rgb_stereo_node.launch using an OAK-D PoE and getting the following error when launching. Is this due to using the PoE camera? 
```
terminate called after throwing an instance of 'nanorpc::core::exception::logic'
  what():  Unexpected Encoding Type
================================================================================
REQUIRED process [rgb_stereo_publisher-3] has died!
process has died [pid 16705, exit code -6, cmd /mnt/wd/aggressionDetection/devel/lib/depthai_examples/rgb_stereo_node __name:=rgb_stereo_publisher __log:=/home/duane/.ros/log/0ad13794-768e-11ed-bcbf-af4e0674c733/rgb_stereo_publisher-3.log].
log file: /home/duane/.ros/log/0ad13794-768e-11ed-bcbf-af4e0674c733/rgb_stereo_publisher-3*.log
Initiating shutdown!
================================================================================
```
The log file also gives this as the only error straight after spin.

Any ideas?

Serafadam, 2022-12-08:
Hi @RhythmDJay, can you check if you still have those errors when running `stereo_intertial_node.launch`? if yes, please run `export DEPTHAI_DEBUG=1` and run it again to get some more info from the logs.

RhythmDJay, 2022-12-08:
Hi @Serafadam, I can run `stereo_inertial_node.launch` I just get this error 
```
Listing available devices...
Device Mx ID: 14442C10915F3FD700
[ERROR] [1670539039.680425422]: PluginlibFactory: The plugin for class 'rviz_imu_plugin/Imu' failed to load.  Error: According to the loaded plugin descriptions the class rviz_imu_plugin/Imu with base class type rviz::Display does not exist. Declared types are  rviz/AccelStamped rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/TwistStamped rviz/WrenchStamped rviz_plugin_tutorials/Imu
[14442C10915F3FD700] [192.168.0.10] [5.293] [SpatialDetectionNetwork(8)] [warning] Network compiled for 6 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
Device USB status: UNKNOWN
```

Serafadam, 2022-12-09:
Hi, seems like you don't have rviz plugins installed:
`sudo apt install ros-noetic-rviz-imu-plugin`

1412kauti, 2022-12-24:
I have the same error....and I already have  `ros-noetic-rviz-imu-plugin` installed , it didnt work !,

any other fixes ?

behold the logs:

```
started roslaunch server http://comet:37723/

SUMMARY
========

PARAMETERS
 * /rgb_stereo_publisher/LRchecktresh: 5
 * /rgb_stereo_publisher/camera_param_uri: package://depthai...
 * /rgb_stereo_publisher/confidence: 200
 * /rgb_stereo_publisher/extended: False
 * /rgb_stereo_publisher/lrcheck: True
 * /rgb_stereo_publisher/subpixel: False
 * /rgb_stereo_publisher/tf_prefix: oak
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES
  /
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rgb_stereo_publisher (depthai_examples/rgb_stereo_node)

auto-starting new master
process[master]: started with pid [2116356]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6d819cda-835e-11ed-a679-032a457d4f74
process[rosout-1]: started with pid [2116379]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [2116386]
process[rgb_stereo_publisher-3]: started with pid [2116387]
[2022-12-24 07:41:51.179] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2022-12-24 07:41:51.179] [debug] Library information - version: 2.19.1, commit:  from , build: 2022-11-28 18:15:09 +0000
[2022-12-24 07:41:51.180] [debug] Initialize - finished
[2022-12-24 07:41:51.235] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.21.tar.xz' open: 1ms, archive read: 54ms
[2022-12-24 07:41:51.396] [debug] Device - OpenVINO version: 2022.1
[2022-12-24 07:41:51.396] [debug] Device - BoardConfig: {"emmc":null,"gpio":[],"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0e b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be 01
[2022-12-24 07:41:51.462] [debug] Resources - Archive 'depthai-device-fwp-adbcc016c8bd5a5580a26d8b6250f77160203666.tar.xz' open: 1ms, archive read: 280ms
[2022-12-24 07:41:52.073] [debug] Searching for booted device: DeviceInfo(name=1.2, mxid=184430103116060F00, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[184430103116060F00] [1.2] [1.153] [system] [warning] PRINT:LeonCss: BootloaderConfig.options1 checksum doesn't match. Is: 0x10304418 should be: 0xFCCBD2D0
GPIO boot mode 0x16, interface USBD
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
initial keepalive, countdown: 10
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
eeprom configuration version: 55AA0006
Reading VERSION 6 --- ->
eeprom configuration load from user area, status: 0
--> brdInit ...
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: ff ff ff
Invalid Flash JEDEC ID... No NOR available
Could not register the spi device
Opening bus for IR driver: 1
Failed to probe IR driver LM3644
Opening bus for IR driver: 2
Failed to probe IR driver LM3644
Opening bus for IR driver: 3
[... 40 lines cut ...]
[184430103116060F00] [1.2] [1.163] [system] [warning] PRINT:LeonCss: I: [Timesync] [   1745680] [XLin] startSync:130     Timesync | Callback not set
[2022-12-24 07:41:53.392] [debug] Device about to be closed...
[2022-12-24 07:41:53.496] [debug] DataOutputQueue (video) closed
[2022-12-24 07:41:53.496] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[2022-12-24 07:41:53.496] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[2022-12-24 07:41:53.496] [debug] DataOutputQueue (depth) closed
[2022-12-24 07:41:53.986] [debug] Watchdog thread exception caught: Couldn't write data to stream: '__watchdog' (X_LINK_ERROR)
[2022-12-24 07:41:54.368] [debug] XLinkResetRemote of linkId: (0)
[2022-12-24 07:41:54.734] [debug] Device closed, 1342
terminate called after throwing an instance of 'nanorpc::core::exception::logic'
  what():  Unexpected Encoding Type
================================================================================REQUIRED process [rgb_stereo_publisher-3] has died!
process has died [pid 2116387, exit code -6, cmd /home/kaito/catkin_ws/devel/lib/depthai_examples/rgb_stereo_node __name:=rgb_stereo_publisher __log:=/home/kaito/.ros/log/6d819cda-835e-11ed-a679-032a457d4f74/rgb_stereo_publisher-3.log].
log file: /home/kaito/.ros/log/6d819cda-835e-11ed-a679-032a457d4f74/rgb_stereo_publisher-3*.log
Initiating shutdown!
================================================================================
[rgb_stereo_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

Serafadam, 2022-12-27:
Hi @1412kauti do you still have those errors when using `stereo_inertial_node` launch?

abegni, 2024-06-07:
> @Serafadam Im currently trying to run depthai_examples/rgb_stereo_node.launch using an OAK-D PoE and getting the following error when launching. Is this due to using the PoE camera?
>
> ```
> terminate [...]

Same error, any ideas?

Serafadam, 2024-06-07:
Hi, could you post more logs according to the [bug template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)? It would be also goo to create a new issue for that.