luxonis/depthai-ros #64: Error during running yolov4_publisher.launch

MoscowskyAnton, 2022-01-31:
Hi! Got new OAK-D-Lite today, must say you guys did a great job!
But faced error when launch yolov4_publisher.launch on ROS Noetic:

Error says:
`Stack trace (most recent call last) in thread 3105:
#5    Object "[", at 0, in nil
#4    Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7f2047feb3bf, in 
#3    Object "/usr/local/lib/libdepthai-core.so", at 0x7f204744c76e, in backward::SignalHandling::sig_handler(int, siginfo_t*, void*)
#2    Object "/usr/local/lib/libdepthai-core.so", at 0x7f204744c67d, in backward::SignalHandling::handleSignal(int, siginfo_t*, void*)
#1    Object "/usr/local/lib/libdepthai-core.so", at 0x7f2047449e5f, in backward::StackTraceImpl<backward::system_tag::linux_tag>::load_here(unsigned long, void*, void*)
#0    Object "/usr/local/lib/libdepthai-core.so", at 0x7f204744cb97, in unsigned long backward::details::unwind<backward::StackTraceImpl<backward::system_tag::linux_tag>::callback>(backward::StackTraceImpl<backward::system_tag::linux_tag>::callback, unsigned long)
Segmentation fault (Address not mapped to object [(nil)])
`
Every launch of yolo ends with this, or with freezing when no image and no data in topic with detected objects.

saching13, 2022-01-31:
Sorry for the trouble. Let me cross-check and get back to you.

daxoft, 2022-02-17:
Hi! I'm trying to reproduce the error on my system, and I couldn't make it. Do you use other OAK-D cameras on that system? Do you use an USB2 port? If so, can you try on USB3? Did you tried to change the cable?

MoscowskyAnton, 2022-02-25:
Hi!
Just tried on different OAK-D-Lite, cable and PC - no such behavior.
Will try again on first one and with different cable. Hope problem is not in device itself.

saching13, 2022-02-25:
Sounds good. Thanks