luxonis/depthai-ros #66: Right side of PoindCloud is lacking.

nisshan-x, 2022-02-08:
Hi,

I bought OAK-D Lite (kickstarter version), installed depthai-ros according to README.md to Ubuntu 20, and launched `roslaunch depthai_examples stereo_nodelet.launch` . But Right side of PoindCloud in Rviz is lacking.

[image]

I tried to run `depthai_demo.py`. Then both left and right side of depth image appeared. I guess OAK-D Lite hardware and USB connection are fine.
[image]

Could you please tell me how to deal with this problem?

nisshan-x, 2022-02-08:
A different angle of view.

[image]

nisshan-x, 2022-02-08:
I tried command with option `roslaunch depthai_examples stereo_nodelet.launch camera_model:=OAK-D-LITE` , but the result was the same...

saching13, 2022-02-08:
I think stereo_nodelet is not configured for OAK-D-Lite with resolution. Will add it today mostly. 
cc: @daxoft  can add this to all the examples here ?

daxoft, 2022-02-15:
Merged into main 
https://github.com/luxonis/depthai-ros-examples/pull/20 
https://github.com/luxonis/depthai-ros/pull/67

closed 2022-02-15 by daxoft