luxonis/depthai-ros #68: CMakeList.txt file in depthai_examples misses install parameters

Skammi, 2022-02-16:
Have added the following in my CMikeList.txt file in the depthai_examples directory

This part added
```
  ## Install these nodes
  install(TARGETS
          crop_control_service
          mobilenet_node
          rgb_node
          rgb_stereo_node
          rgb_subscriber_node
          # stereo_mimic_node
          stereo_inertial_node
          stereo_node
          yolov4_spatial_node
          RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
  )

  install(DIRECTORY launch/
	DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/launch
	PATTERN ".svn" EXCLUDE
  )
```
before endif
```
endif()
```

saching13, 2022-02-16:
Oh yes. Thanks. Feel free to create a PR.

Skammi, 2022-02-24:
Done in depthai-ros-examples pull request #23

saching13, 2022-02-24:
Thanks for the contribution.

closed 2022-02-24 by saching13