luxonis/depthai-ros #69: stereo_publisher has died when I run stereo_nodelet.launch with OAK-D Lite

nisshan-x, 2022-02-16:
Hi,

I tried to run `roslaunch depthai_examples stereo_nodelet.launch`, but process has died.

<details>
<summary>Error message</summary>

```
$ roslaunch depthai_examples stereo_nodelet.launch
... logging to /home/xxx/.ros/log/a35c3a4a-8ee5-11ec-a5b1-670792c734cc/roslaunch-ubuntu-66757.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:45569/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /stereo_publisher/LRchecktresh: 5
 * /stereo_publisher/camera_param_uri: package://depthai...
 * /stereo_publisher/confidence: 200
 * /stereo_publisher/extended: False
 * /stereo_publisher/lrcheck: True
 * /stereo_publisher/mode: depth
 * /stereo_publisher/subpixel: True
 * /stereo_publisher/tf_prefix: oak

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_pointcloud_nodelet (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    stereo_publisher (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [66771]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a35c3a4a-8ee5-11ec-a5b1-670792c734cc
process[rosout-1]: started with pid [66783]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [66789]
process[nodelet_manager-3]: started with pid [66791]
process[stereo_publisher-4]: started with pid [66792]
process[depth_image_convertion_nodelet-5]: started with pid [66797]
process[depth_image_to_pointcloud_nodelet-6]: started with pid [66800]
[ INFO] [1644987749.408312651]: Loading nodelet /stereo_publisher of type depthai_examples/StereoNodelet to manager nodelet_manager with the following remappings:
process[rviz-7]: started with pid [66811]
[ INFO] [1644987749.413470156]: Initializing nodelet with 3 worker threads.
 Bad parameters -> 1
[FATAL] [1644987749.503596763]: Failed to load nodelet '/stereo_publisher` of type `depthai_examples/StereoNodelet` to manager `nodelet_manager'
================================================================================REQUIRED process [stereo_publisher-4] has died!
process has died [pid 66792, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load depthai_examples/StereoNodelet nodelet_manager __name:=stereo_publisher __log:=/home/xxx/.ros/log/a35c3a4a-8ee5-11ec-a5b1-670792c734cc/stereo_publisher-4.log].
log file: /home/xxx/.ros/log/a35c3a4a-8ee5-11ec-a5b1-670792c734cc/stereo_publisher-4*.log
Initiating shutdown!
================================================================================
[rviz-7] killing on exit
[depth_image_to_pointcloud_nodelet-6] killing on exit
[stereo_publisher-4] killing on exit
[nodelet_manager-3] killing on exit
[oak_state_publisher-2] killing on exit
[depth_image_convertion_nodelet-5] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

</details>

Also I tried `roslaunch depthai_examples stereo_node.launch`, it was ok. So I don't think the cause is my environment.

saching13, 2022-02-16:
Hello,
Sorry for the issue. Looks like @daxoft missed a parameter setting for that example. Just fixed it and updated it.

nisshan-x, 2022-02-16:
It works well. Thank you very much!

closed 2022-02-16 by nisshan-x