luxonis/depthai-ros #71: depthai_bridge/BridgePublisher::~BridgePublisher()  Bug?

tosa-no-onchan, 2022-02-19:
stereo_nodelet.cpp terminates with error message.
I think, 
depthai_bridge/BridgePublisher::~BridgePublisher()  Bug?

Are try and catch codes needed?

https://github.com/luxonis/depthai-ros/blob/21baff8e923e7ea0df60a2138174e5d14d306fdd/depthai_bridge/include/depthai_bridge/BridgePublisher.hpp#L163-L167
----
template <class RosMsg, class SimMsg> 
BridgePublisher<RosMsg, SimMsg>::~BridgePublisher(){
  try{
    _readingThread.join();  
  }
  catch(_Unwind_RaiseException e){}
}

themarpe, 2022-02-21:
@tosa-no-onchan thanks for reporting!
Do you have a specific scenario where/when this happens in that example?

@saching13 I think the following is required:
```
template <class RosMsg, class SimMsg>
BridgePublisher<RosMsg, SimMsg>::~BridgePublisher() {
    if(_readingThread.joinable()) _readingThread.join();
}
```

As if `startPublisherThread` isn't called, the above will throw

saching13, 2022-02-21:
Yes. Sorry I missed that. Fixing it.

saching13, 2022-02-21:
Fixed it.

tosa-no-onchan, 2022-02-22:
Thank you,seaching13.

It worked!
Stereo_nodelet.launch become to terminate with no error.

closed 2022-02-22 by saching13