luxonis/depthai-ros #78: Error in executing the command "sudo wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/main/install_dependencies.sh | sudo bash"

sameertodkar, 2022-02-23:
System: Ubuntu 18, ROS Melodic
[image]

sameertodkar, 2022-02-23:
On executing the following line, the above error is displayed.
`cmake --build depthai-core/build --target install`

There are certain tests that fail while executing the install depedencies bash file.
[image]

saching13, 2022-02-23:
Any idea @themarpe  ?

themarpe, 2022-02-23:
The tests for flags are just checks - could've been better named...

Seems like C++14 supports is missing... @sameertodkar can you check compiler version you are using? 
```
g++ --version
```

Seems like its version 5 by the looks of the paths.. Might need to update that or change which compiler is used...

sameertodkar, 2022-02-23:
@themarpe  I tested it and following are the results.
`g++ (Ubuntu 5.5.0-12ubuntu1) 5.5.0 20171010
Copyright (C) 2015 Free Software Foundation, Inc.
`
Yeah it is using the version 5.

themarpe, 2022-02-23:
@sameertodkar 
```
sudo apt install build-essential
```

should install GCC 7.5.0 on Ubuntu 18.04.
Note, this might change default compiler, make sure that it doesn't have any bad consequences in your setup.

And check if an alternative is set:
```
update-alternatives --query gcc
```

sameertodkar, 2022-02-23:
@themarpe 
Interesting stuff, on running the above command 
`sudo apt install build-essential`
I get the following output
`Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.4ubuntu1).`

It does not update the gcc. 

The command `update-alternatives --query gcc`
gives the following results:

`Name: gcc
Link: /usr/bin/gcc
Slaves:
 g++ /usr/bin/g++
Status: manual
Best: /usr/bin/gcc-5
Value: /usr/bin/gcc-5

Alternative: /usr/bin/gcc-5
Priority: 60
Slaves:
 g++ /usr/bin/g++-5
`

themarpe, 2022-02-23:
@sameertodkar 
Can you try running this then:
```
sudo apt install gcc-7
```

And rerunning `update-alternatives --query gcc` if its set as default now

Afterwards, try running he initial command again

sameertodkar, 2022-02-23:
@themarpe I tried that, there is gcc-7 in my system although it does not show up in the list when you run the command.
` sudo update-alternatives --config gcc`
It gives out:
There is 1 choice for the alternative gcc (providing /usr/bin/gcc).

  Selection    Path            Priority   Status
------------------------------------------------------------
  0            /usr/bin/gcc-5   60        auto mode
* 1            /usr/bin/gcc-5   60        manual mode

I will try reinstalling gcc and check.

themarpe, 2022-02-23:
Can you check if GCC-7 is installed
```
g++-7 --version
```

If so, I think you could also do the following (YMMV):

```
CC=gcc-7 CXX=g++-7 sudo wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/main/install_dependencies.sh | sudo bash
```

The above sets compilers to use for compiling C and C++ sources

sameertodkar, 2022-02-23:
The issue is resolved, I executed the following commands:

`sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt-get install gcc-7 g++-7
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 60 --slave /usr/bin/g++ g++ /usr/bin/g++-7
`
Then tested the first command it worked with no errors. I will close the issue!
Thanks @themarpe

closed 2022-02-23 by sameertodkar