luxonis/depthai-ros #79: Generating pointcloud data

sameertodkar, 2022-02-23:
@saching13 I am trying to publish the pointcloud data using the oak-d pro. I am facing the following issues.

1. Is it possible to publish the pointcloud data using the depthai-ros and OAK-D pro? I am looking for the pointcloud data which has the message type sensor_msgs/PointCloud1 or PointCloud2.

http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/PointCloud.html 

2. On adding other topics to RViz, it crashes with a message on the terminal 
`QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
`
I think it may not be related. I have attached the image.

[image]

3. On trying out the rgb_stereo_node.launch, it does not publish any data on to the topic as none of it can be visualized.

[image]

saching13, 2022-02-23:
Use the `stereo_nodelet.launch` file for pointcloud.

rafaelRinconFo, 2022-03-24:
> Use the `stereo_nodelet.launch` file for pointcloud.

Hi! What would be the case for obtaining the pointcloud in ROS2?

saching13, 2022-03-24:
`ros2 launch  depthai_examples stereo_inertial_node.launch.py`
&
`ros2 launch  depthai_examples stereo.launch.py`
Both works.

saching13, 2022-03-24:
For ROS1: 
`roslaunch depthai_examples stereo_inertial_node.launch`

Serafadam, 2022-11-07:
Closing due to inactivity, if you think this should remain open, please tag me in a comment here.

closed 2022-11-07 by Serafadam