luxonis/depthai-ros #82: python-rosdep2 is dangerous

Russ76, 2022-02-26:
This program is listed in your install instructions for Melodic ROS. It removed most of my ROS installation! I found that python-rosdep worked fine. The ROS wiki says that python-rosdep2 is unstable. 

Compiling on a Jetson TX1 I found that I had to command "make -j1 -l4" otherwise it would freeze up at 67% completed.

themarpe, 2022-03-01:
Sorry for the issues with caused by erroneous installation instructions. I'll hand this over to @saching13 to check and properly guard against this happening in the future.

saching13, 2022-03-01:
Sorry for the issue. I will add Optional instructions to use the same where needed.

mikkeljakobsen, 2022-03-15:
I did not manage to get it working with ROS melodic and python-rosdep. It worked with python-rosdep2, but it did remove most of my ROS installation as you say.

saching13, 2022-03-15:
> I did not manage to get it working with ROS melodic and python-rosdep. It worked with python-rosdep2, but it did remove most of my ROS installation as you say.

is this specific to jetson platform? I didn't see issues on the usual Ubuntu 18.04.

mikkeljakobsen, 2022-03-16:
> > I did not manage to get it working with ROS melodic and python-rosdep. It worked with python-rosdep2, but it did remove most of my ROS installation as you say.
> 
> is this specific to jetson platform? I didn't see issues on the usual Ubuntu 18.04.

No, it's in a docker container on an amd64 machine.

mikkeljakobsen, 2022-03-17:
I solved the issue by running `ldconfig`. Apparently this was not needed when using python-rosdep2.

Serafadam, 2022-11-09:
Closing due to inactivity, if you think this should remain open, please tag me in a comment here.

closed 2022-11-09 by Serafadam