luxonis/depthai-ros #87: problem while running ros2 launch depthai_examle yolov4_publisher_launch.py

Nicoladao, 2022-03-03:
[image]
Hi,
    When i run ros2 launch depthai_examle yolov4_publisher_launch.py, and it cannot continue.Do you know what problem with it ,how to fix?
Thank you!
ubuntu20.04 OAK-D-POE ROS2

saching13, 2022-03-03:
Actually it is working for you. I have not added visualization for it. Start an Rviz or Rqt_image_view to see visualization. 
That said we still have to add plugins for our custom messages too.

Nicoladao, 2022-03-08:
> Actually I think it is working for you. I have not added visualization for it. Start an Rviz or Rqt_image_view to see visualization. That said we still have to add plugins for our custom messages too.

Thanks, that's true. I can echo the topic.I thought there was a visualization.

And,if this package can use the function of getCvFrame() and getFrame()?

Nicoladao, 2022-03-08:
> Actually I think it is working for you. I have not added visualization for it. Start an Rviz or Rqt_image_view to see visualization. That said we still have to add plugins for our custom messages too.

[image]
the error show like this.

It is foxy and I create my own ros-package to run the yolo-net.Should i set opencv in cmakelist.txt?

saching13, 2022-03-09:
I think you need to add `depthai::opencv` as dependency here.
[here](https://github.com/luxonis/depthai-core#cmake) is more documentation on the same.

Nicoladao, 2022-03-09:
> I think you need to add `depthai::opencv` as dependency here. And also make sure you have OpenCV installed so that those functions are built using that. [here](https://github.com/luxonis/depthai-core#cmake) is more documentation on the same.

Thanks very much.I forget to add depthai::opencv as dependency, it is works now!
very appreciated

closed 2022-03-09 by Nicoladao