luxonis/depthai-ros #89: use ros2 and run examples oak-d ascension cannot connect

sunil-gitcode, 2022-03-10:
cmd :ros2 launch depthai_examples stereo.launch.py
[image]
[image]

saching13, 2022-03-10:
Hello,
Sorry fro the issue.

1. Can you check if the stereo works on depthai_demo.py ? 
2. Can you cross-check if the right DepthAI C++ library is installed.

sunil-gitcode, 2022-03-11:
Run python3 depthai-demo.py but apparently it's fine so I don't know what the problem is  
[image]

saching13, 2022-03-11:
In ros installation steps did you install the main branch ? And also when installing the dependencies was there any change ?

sunil-gitcode, 2022-03-11:
I installed it according to the tutorial,command is sudo apt install ros-galactic-desktop
.Dependencies are installed automatically,I wonder what has changed about it.

saching13, 2022-03-11:
Can you run the Installation dependencies [here](https://github.com/luxonis/depthai-ros#install-dependencies) followed by cleaning the WS and rebuilding the packages?

sunil-gitcode, 2022-03-11:
[image]

Follow the depth-ros tutorial ,i coclon build it and It say that the compilation succeeded,.Run to this step source devel/setup.bash,it  says no such file,so i put it into the ros2 project and compile it again.

saching13, 2022-03-11:
`devel/setup.bash` is for ros 1. `install/setup.bash` is for ros2.

sunil-gitcode, 2022-03-11:
Thank you for your help
1.It is depthai-ros project,It's just that the name is wrong.
2.depth-ros Dependencies are perfectly installed.
2. According to the tutorial, Dependency installation is based on install_dependencies.sh，Its compile and install prompts are successful.
3. Do I need to try all the steps again including ro2 installation

saching13, 2022-03-11:
And you are still facing the issue ? 
3. No.

sunil-gitcode, 2022-03-11:
yes,I still don't know what his problem is.I have to put the point cloud data into ROS2，It's an engineering project.

saching13, 2022-03-11:
> put

Can you share the full logs on the terminal ?

sunil-gitcode, 2022-03-11:
[log.txt](https://github.com/luxonis/depthai-ros/files/8228323/log.txt)
This is the log

saching13, 2022-03-11:
Ok. In depthai_demo.py can you share the left camera image on the display ?

saching13, 2022-03-11:
And logs from depthai demo too.

sunil-gitcode, 2022-03-11:
[image]
Is that the choice

saching13, 2022-03-11:
[image]
No. Once the demo starts you have to change the camera where I have marked to left. And also share the logs printed in the terminal when running the demo

sunil-gitcode, 2022-03-11:
[image]
It doesn't seem to have that option

saching13, 2022-03-11:
that means like Erik mentioned the connector might be disconnected or you have OAK-1. 
I think your connector might be disconnected.

saching13, 2022-03-11:
Yes. Or we can get it replaced I think. Do you have an idea when it was purchased ?

sunil-gitcode, 2022-03-11:
I bought it in December 21, I tried to repair it first to see.

sunil-gitcode, 2022-03-11:
thanks you for your help, The problem was solved.

closed 2022-03-11 by saching13