luxonis/depthai-ros #92: oak-d-lite with vins-gpu

sumzora, 2022-03-21:
When I run roslaunch vins oak.launch according to this website (https://www.arducam.com/docs/opencv-ai-kit-oak/performing-location-with-visual-slam/), the error comes with **unused args [camera_name],for include ~<path>/luxonis/depthai-ros/depthai_bridge/launch/urdf.launch**
what's the camera name?
Thank you very much!

saching13, 2022-03-21:
replace the `camera_name` with `tf_prefix`

sumzora, 2022-03-22:
Oh, I still can not fix it. The specific error is following:
[image]
But my depthai-ros is installed well since I can run roslaunch depthai_examples stereo_node.launch and get pointcloud.

Serafadam, 2022-11-09:
Hi, wanted to check in if you still have problems with that, the repo has been updated a little bit since March (and will be overhauled in near future), I'll close this issue for now since it's inactive for now but we can reopen it if you feel this still needs to be addressed.

closed 2022-11-09 by Serafadam