luxonis/depthai-ros #95: Run stereo.launch.py but pointcloud2 frame rate is very low

sunil-gitcode, 2022-03-30:
Use the default demo and run stereo.launch.py, but the pointCloud2 frame rate on RVIz looks pretty low at about one frame per second

sunil-gitcode, 2022-03-31:
> Can I know which device is used here and what host ?

device oak-d and host is laptop computer

saching13, 2022-03-31:
Ok. Can you do `rostopic hz` test on the depth message and the pointcloud topics and share the screenshot with us please? 
And I would suggest switching to `stereo_inertial_node` since it has lot of improvements.

sunil-gitcode, 2022-03-31:
> Ok. Can you do `rostopic hz` test on the depth message and the pointcloud topics and share the screenshot with us please? And I would suggest switching to `stereo_inertial_node` since it has lot of improvements.
This use stereo_inertial_node  and screenshot 
[image]

[image]

saching13, 2022-03-31:
What about depth image frame rate ? 
And just to cross confirm. The repo is up-to date right ?

saching13, 2022-03-31:
Another thing you could share is resource consumption using `top`

sunil-gitcode, 2022-03-31:
> What about depth image frame rate ? And just to cross confirm. The repo is up-to date right ?

1.[image]
2.yes
3.
[image]
4.oak-d

saching13, 2022-03-31:
Thanks for the details. Can I know which ROS version this is ?

sunil-gitcode, 2022-04-01:
> Thanks for the details. Can I know which ROS version this is ?
Ros version is ros galactic

saching13, 2022-04-04:
[image]
From `stereo.launch.py`. I don't see an issue on mine. 
Things to check.
1. DDS connection.
2. Bandwidth available

sunil-gitcode, 2022-05-09:
I have tried many times and it is still the same. How should I check it?

Luxonis-Brandon, 2022-05-11:
@daxoft may be able to advise.

daxoft, 2022-05-11:
I haven't been able to reproduce the bug, do you have at your disposal another camera and/or host? Does the same issue occur with other examples as well?

sunil-gitcode, 2022-05-11:
> I haven't been able to reproduce the bug, do you have at your disposal another camera and/or host? Does the same issue occur with other examples as well?

I am using other computers as well, it is worth noting that they are all running ubantu20 on virtual machines

sunil-gitcode, 2022-05-12:
> thanks, VM might affect it, I will try it on VMware also; might not matter too much but are you running VMware from linux or windows? And does the host VM has ROS2 installed as well?

1.In the Windows
2.yes

vinnnyr, 2022-10-21:
Is this using a `reliable` publisher? You might be facing issues if that's the case if the point cloud is sufficiently big. 

Another thing -- what RMW do you use? If it's galactic the default is cyclonedds, and I have observed that without configuration sometimes cyclonedds may try to multicast large messages.

Serafadam, 2022-11-09:
By default `depth_image_proc` uses Best Effort policy. There's a lot of parameters that influence framerate of such big messages in ROS such as DDS configuration, network speed and some internal linux configs, we'll probably add some DDS configurations to the examples, but for now you can check out https://docs.ros.org/en/humble/How-To-Guides/DDS-tuning.html

rickarmstrong, 2022-12-13:
FYI I just ran into this issue last night with my OAK-D Pro, W, over USB 3. Dell laptop running `Ubuntu 20.04.5 LTS (GNU/Linux 5.15.0-56-generic x86_64)` , ROS 2 Galactic, CycloneDDS. Switched to `RMW_IMPLEMENTATION=rmw_fastrtps_cpp` and problem disappeared.

vinnnyr, 2022-12-13:
If you want to restrict CycloneDDS to only use multicast for discovery (and not for message transmission) you can set your config xml to something like this:
```
<?xml version="1.0" encoding="UTF-8" ?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/iceoryx/etc/cyclonedds.xsd">
    <Domain id="any">
        <General>
            <AllowMulticast>spdp</AllowMulticast>
        </General>
    </Domain>
</CycloneDDS>
```
This has helped us for large messages such as pointclouds on our cyclonedds config.

closed 2023-09-14 by Serafadam