luxonis/depthai-ros #96: URDF describes IMU frames for non-imu camera, resulting in "'imu_offset_x' is not defined" error

AlessioMorale, 2022-04-03:
Running depthai_descr.urdf.xacro setting `camera_model:=OAK-D-LITE` fails with the following error:

```
name 'imu_offset_x' is not defined 
when evaluating expression 'imu_offset_x'
when instantiating macro: depthai_camera (/home/alessio/code/ros/depthai/src/luxonis/depthai-ros/depthai_bridge/urdf/include/depthai_macro.urdf.xacro)
```
Steps to reproduce:

- setup an empty ros workspace as described in [README](https://github.com/luxonis/depthai-ros/blob/main/README.md) 
- catkin build
- catkin source
- roslaunch depthai_bridge urdf.launch camera_model:=OAK-D-LITE # immediately fails with the error above

saching13, 2022-04-03:
Oh my bad. I will add the IMU position for the rest.

AlessioMorale, 2022-04-03:
I fixed this with #97 by enabling IMU related frames using the `has_imu` property. Currently only for OAK-D (that's the only one defining imu_* variables).

saching13, 2022-04-04:
Actually post kickstarted verision of OAK-D-Lite has IMU. Ididn't get time to add it here yet. But I agree. I will add your PR for now and upgrade it later. 
THanks for the PR.

PR #97 (2022-04-04): fix #96: Only Set up IMU frames for cameras with IMU
fixes #96:
Only include IMU frames for camera with IMU
commit c94cab97 (2022-04-04): fix #96: Set up IMU frames only for cameras with IMU (#97)
commit 8bb209fe (2022-04-05): Devel (#99)

* Added workflow on devel too

* Timestamp update (#80)

* Updated depthai to use timestamp from 2.15 of DepthAI

* Upgrade version:

* Add depthai version check for upgrade

* Imu interpolation (#90)

* Added workflow on devel too (#81)

* Fix interpolation

* WIP updated Bridge publisher

* Devel (#83)

* Added workflow on devel too

* Timestamp update (#80)

* Updated depthai to use timestamp from 2.15 of DepthAI

* Upgrade version:

* Add depthai version check for upgrade

* Refactored Bridge Publisher, Image Converter and Disparity converter

* Updated Image Detection and Spa [... cut]
commit b2f52781 (2022-04-15): Daxoft/fix disparity (#101)

* Devel (#99)

* Added workflow on devel too

* Timestamp update (#80)

* Updated depthai to use timestamp from 2.15 of DepthAI

* Upgrade version:

* Add depthai version check for upgrade

* Imu interpolation (#90)

* Added workflow on devel too (#81)

* Fix interpolation

* WIP updated Bridge publisher

* Devel (#83)

* Added workflow on devel too

* Timestamp update (#80)

* Updated depthai to use timestamp from 2.15 of DepthAI

* Upgrade version:

* Add depthai version check for upgrade

* Refactored Bridge Publisher, Image Converter and Disparity converter

*  [... cut]
closed 2022-04-05 by saching13