luxonis/depthai-ros #98: URDF does not provide any of the camera attachment

AlessioMorale, 2022-04-03:
The camera URDF is missing any attachment link, making it complex to attach it to a robot model.

For example, OAK-D-LITE is missing frames for:
- bottom 1/4 UNC Screw attachment
- rear M4 attachments.

saching13, 2022-04-03:
Do you mean it is missing from the model files or the URDF link doesn't exist ?

AlessioMorale, 2022-04-03:
There are no links defined for those attachments in the URDF.
Just to provide some background: Designing the robot mechanics, I know exactly the placement of every camera attachment, thus I can use this information to precisely integrate the camera URDF with the rest of the robot description.
Look at the Realsense URDF for an example: https://github.com/IntelRealSense/realsense-ros/blob/e4938bbbfd7b285fe71f81fe69cebcc76dbf72fc/realsense2_description/urdf/_d435.urdf.xacro#L51