luxonis/depthai-ros #104: Color Point Cloud Support

braca51e, 2022-04-27:
Is there anyway to get a color point cloud, currently just overlaying the right camera frame gives a monochrome cloud, so why is there no demo for color point cloud?

Thanks

Luxonis-Brandon, 2022-04-27:
Yes there is a way but I'm not technical enough to share.  @OanaMariaVatavu I think can share but might need help from @daxoft on the ROS side.

saching13, 2022-04-27:
Hello @braca51e ,
      You can see RGB pointCloud by running `roslaunch depthai_examples stereo_inertial_node.launch`
 

[image]

braca51e, 2022-04-27:
@saching13 @Luxonis-Brandon Thanks for your answers will test!

braca51e, 2022-04-27:
@saching13 I've tested it and it works but after some frames it freezes out! Are you experiencing the same problem?

saching13, 2022-04-27:
ran it for 10min. don't see an issue. do you get any error ? can I know which host and ROS version?

braca51e, 2022-04-27:
no errors I'm using noetic

saching13, 2022-04-27:
Hhmm. Same here. 
Which OAK version ?

braca51e, 2022-04-27:
OAK-D no poe

braca51e, 2022-04-27:
For some reason it stops publishing the points after a few seconds

saching13, 2022-04-27:
WIll check on OAK-D-Poe and get back on this

saching13, 2022-05-04:
Ah got it. queue size of depthai or of pointcloud nodelet ?

closed 2022-05-04 by saching13