luxonis/depthai-ros #106: Compiling OpenCV message

FPSychotic, 2022-04-29:
```
-- +++ processing catkin package: 'depthai_ros_msgs'
-- ==> add_subdirectory(luxonis/depthai-ros/depthai_ros_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- depthai_ros_msgs: 5 messages, 2 services
-- +++ processing catkin package: 'depthai_bridge'
-- ==> add_subdirectory(luxonis/depthai-ros/depthai_bridge)
-- ----------------Did not find OpenCV 4, trying OpenCV 3--------------
-- 2.15.3
-- Found Boost: /usr/lib/aarch64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")  
-- ------------------------------------------
-- Depthai Bridge is being built using CATKIN.
-- ------------------------------------------

```

I receive this message having system Opencv 4.5.5 and ROS opencv 4.2. 
with cv.2__version__ it says 4.5.5
I tried several things , and I will appear always.

serna01, 2022-05-16:
I am having the same issue:

```
-- +++ processing catkin package: 'depthai_ros_msgs'
-- ==> add_subdirectory(luxonis/depthai-ros/depthai_ros_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- depthai_ros_msgs: 5 messages, 2 services
-- +++ processing catkin package: 'depthai_bridge'
-- ==> add_subdirectory(luxonis/depthai-ros/depthai_bridge)
-- ----------------Did not find OpenCV 4, trying OpenCV 3--------------
CMake Error at luxonis/depthai-ros/depthai_bridge/CMakeLists.txt:25 (find_package):
  Could not find a package configuration file provided by "depthai" with any
  of the following names:

    depthaiConfig.cmake
    depthai-config.cmake

  Add the installation prefix of "depthai" to CMAKE_PREFIX_PATH or set
  "depthai_DIR" to a directory containing one of the above files.  If
  "depthai" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/icra/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/icra/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:3820: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```
Ubuntu 18.04
ROS melodic
OpenCV : 4.5.5

saching13, 2022-05-17:
You are not able to find the depthai-core library using find package. Please check if it's installed.

serna01, 2022-05-17:
How can I check if it's installed? I followed all the steps in the installation procedure.
Is there an additional command to install the depthai-core library?

serna01, 2022-05-17:
I did manage to install the depthai-core library, now when i am on step 7 of setting up procedure (catkin make), the following errors appear:
``
CMake Warning at mapviz/mapviz/CMakeLists.txt:133 (add_library):
  Cannot generate a safe runtime search path for target rqt_mapviz because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libQt5Concurrent.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5OpenGL.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5Widgets.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libuuid.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5Gui.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5Core.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at mapviz/mapviz/CMakeLists.txt:162 (add_executable):
  Cannot generate a safe runtime search path for target mapviz because files
  in some directories may conflict with libraries in implicit directories:

    runtime library [libQt5Concurrent.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5OpenGL.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5Widgets.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libuuid.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5Gui.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5Core.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at mapviz/multires_image/CMakeLists.txt:74 (add_library):
  Cannot generate a safe runtime search path for target multires_image
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libuuid.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5OpenGL.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5Widgets.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5Gui.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5Concurrent.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libQt5Core.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at mapviz/multires_image/CMakeLists.txt:100 (add_library):
  Cannot generate a safe runtime search path for target multires_widget
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libQt5Concurrent.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
    runtime library [libuuid.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/icra/miniconda3/lib
[... 898 lines cut ...]
//usr/lib/libgdal.so.20: undefined reference to `TIFFSwabArrayOfDouble@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSetSubDirectory@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFUnlinkDirectory@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFSetField@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFCurrentDirOffset@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFIsCODECConfigured@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFUnsetField@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFDataWidth@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFRewriteDirectory@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFGetSizeProc@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSetDirectory@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFReadScanline@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFNumberOfTiles@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
luxonis/depthai-ros-examples/depthai_examples/CMakeFiles/rgb_stereo_node.dir/build.make:140: recipe for target '/home/icra/catkin_ws/devel/lib/depthai_examples/rgb_stereo_node' failed
make[2]: *** [/home/icra/catkin_ws/devel/lib/depthai_examples/rgb_stereo_node] Error 1
CMakeFiles/Makefile2:10784: recipe for target 'luxonis/depthai-ros-examples/depthai_examples/CMakeFiles/rgb_stereo_node.dir/all' failed
make[1]: *** [luxonis/depthai-ros-examples/depthai_examples/CMakeFiles/rgb_stereo_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
``

FPSychotic, 2022-05-17:
Try use catkin_make -j4

On Tue, 17 May 2022, 21:04 Alejandro Serna M, ***@***.***>
wrote:

> I did manage to install the depthai-core library, now when i am on step 7
> of setting up procedure (catkin make), the following errors appear:
> ``
> CMake Warning at mapviz/mapviz/CMakeLists.txt:133 (add_library):
> Cannot generate a safe runtime search path for target rqt_mapviz because
> files in some directories may conflict with libraries in implicit
> directories:
>
> runtime library [libQt5Concurrent.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5OpenGL.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5Widgets.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libuuid.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5Gui.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5Core.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
>
> Some of these libraries may not be found correctly.
>
> CMake Warning at mapviz/mapviz/CMakeLists.txt:162 (add_executable):
> Cannot generate a safe runtime search path for target mapviz because files
> in some directories may conflict with libraries in implicit directories:
>
> runtime library [libQt5Concurrent.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5OpenGL.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5Widgets.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libuuid.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5Gui.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5Core.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
>
> Some of these libraries may not be found correctly.
>
> CMake Warning at mapviz/multires_image/CMakeLists.txt:74 (add_library):
> Cannot generate a safe runtime search path for target multires_image
> because files in some directories may conflict with libraries in implicit
> directories:
>
> runtime library [libuuid.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5OpenGL.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5Widgets.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5Gui.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5Concurrent.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
> runtime library [libQt5Core.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
>   /home/icra/miniconda3/lib
>
> Some of these libraries may not be found correctly.
>
> CMake Warning at mapviz/multires_image/CMakeLists.txt:100 (add_library):
> Cannot generate a safe runtime search path for target multires_widget
> because files in some directories may conflict with libraries in implicit
[... 1211 lines cut ...]
> //usr/lib/libgdal.so.20: undefined reference to
> ***@***.***_4.0'
> //usr/lib/libgdal.so.20: undefined reference to
> ***@***.***_4.0' //usr/lib/libgdal.so.20: undefined
> reference to ***@***.***_4.0'
> //usr/lib/libgdal.so.20: undefined reference to
> ***@***.***_4.0' //usr/lib/x86_64-linux-gnu/libpoppler.so.73:
> undefined reference to ***@***.***_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined
> reference to ***@***.***_4.0' //usr/lib/libgdal.so.20:
> undefined reference to ***@***.***_4.0'
> //usr/lib/libgdal.so.20: undefined reference to ***@***.***_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference
> to ***@***.***_4.0'
> //usr/lib/libgdal.so.20: undefined reference to ***@***.***_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference
> to ***@***.***_4.0'
> //usr/lib/libgdal.so.20: undefined reference to
> ***@***.***_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference
> to ***@***.***_4.0'
> /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined
> reference to ***@***.***_4.0'
> //usr/lib/x86_64-linux-gnu/libgeotiff.so.2: undefined reference to
> ***@***.*

serna01, 2022-05-17:
catkin_make -j4 : 

`Base path: /home/icra/catkin_ws
Source space: /home/icra/catkin_ws/src
Build space: /home/icra/catkin_ws/build
Devel space: /home/icra/catkin_ws/devel
Install space: /home/icra/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/icra/catkin_ws/build"
####
####
#### Running command: "make -j4" in "/home/icra/catkin_ws/build"
####
[  0%] Built target actionlib_generate_messages_cpp
[  1%] Built target rplidarNodeClient
[  4%] Built target rplidarNode
[  4%] Built target std_msgs_generate_messages_lisp
[  4%] Built target geometry_msgs_generate_messages_cpp
[  4%] Built target std_msgs_generate_messages_nodejs
[  7%] Built target geometric_shapes
[  7%] Built target geometry_msgs_generate_messages_eus
[  7%] Built target geometry_msgs_generate_messages_lisp
[  7%] Built target std_msgs_generate_messages_eus
[  7%] Built target geometry_msgs_generate_messages_nodejs
[  7%] Built target std_msgs_generate_messages_cpp
[  7%] Built target geometry_msgs_generate_messages_py
[  7%] Built target std_msgs_generate_messages_py
[  7%] Built target roscpp_generate_messages_py
[  7%] Built target roscpp_generate_messages_nodejs
[  7%] Built target roscpp_generate_messages_eus
[  7%] Built target roscpp_generate_messages_lisp
[  7%] Built target sensor_msgs_generate_messages_py
[  7%] Built target rosgraph_msgs_generate_messages_cpp
[  7%] Built target rosgraph_msgs_generate_messages_eus
[  7%] Built target roscpp_generate_messages_cpp
[  7%] Built target sensor_msgs_generate_messages_nodejs
[  7%] Built target rosgraph_msgs_generate_messages_lisp
[  7%] Built target rosgraph_msgs_generate_messages_nodejs
[  7%] Built target rosgraph_msgs_generate_messages_py
[  7%] Built target sensor_msgs_generate_messages_cpp
[  7%] Built target sensor_msgs_generate_messages_eus
[  7%] Built target sensor_msgs_generate_messages_lisp
[  7%] Built target marti_common_msgs_generate_messages_cpp
[  7%] Built target topic_tools_generate_messages_nodejs
[  7%] Built target topic_tools_generate_messages_lisp
[  7%] Built target topic_tools_generate_messages_eus
[  7%] Built target tf2_msgs_generate_messages_py
[  7%] Built target tf2_msgs_generate_messages_cpp
[  7%] Built target topic_tools_generate_messages_cpp
[  7%] Built target _mapviz_generate_messages_check_deps_AddMapvizDisplay
[  7%] Built target uuid_msgs_generate_messages_cpp
[  7%] Built target actionlib_generate_messages_lisp
[  7%] Built target actionlib_generate_messages_py
[  7%] Built target geographic_msgs_generate_messages_py
[  7%] Built target diagnostic_msgs_generate_messages_py
[  7%] Built target uuid_msgs_generate_messages_eus
[  7%] Built target nodelet_generate_messages_eus
[  7%] Built target geographic_msgs_generate_messages_cpp
[  7%] Built target dynamic_reconfigure_generate_messages_eus
[  7%] Built target diagnostic_msgs_generate_messages_lisp
[  7%] Built target dynamic_reconfigure_generate_messages_lisp
[  7%] Built target geographic_msgs_generate_messages_nodejs
[  7%] Built target bond_generate_messages_py
[  7%] Built target std_srvs_generate_messages_nodejs
[  7%] Built target std_srvs_generate_messages_py
[  7%] Built target std_srvs_generate_messages_cpp
[  7%] Built target diagnostic_msgs_generate_messages_eus
[  7%] Built target swri_roscpp_generate_messages_lisp
[  7%] Built target geographic_msgs_generate_messages_eus
[  7%] Built target actionlib_msgs_generate_messages_lisp
[... 385 lines cut ...]
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFOpen@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSwabArrayOfDouble@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSetSubDirectory@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFUnlinkDirectory@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFSetField@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFCurrentDirOffset@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFIsCODECConfigured@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFUnsetField@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFDataWidth@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFRewriteDirectory@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFGetSizeProc@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSetDirectory@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFReadScanline@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFNumberOfTiles@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
luxonis/depthai-ros-examples/depthai_examples/CMakeFiles/stereo_inertial_node.dir/build.make:140: recipe for target '/home/icra/catkin_ws/devel/lib/depthai_examples/stereo_inertial_node' failed
make[2]: *** [/home/icra/catkin_ws/devel/lib/depthai_examples/stereo_inertial_node] Error 1
CMakeFiles/Makefile2:10436: recipe for target 'luxonis/depthai-ros-examples/depthai_examples/CMakeFiles/stereo_inertial_node.dir/all' failed
make[1]: *** [luxonis/depthai-ros-examples/depthai_examples/CMakeFiles/stereo_inertial_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4" failed`

an99990, 2022-10-21:
I am also encountering these warnings during catkin_make. This is related to opencv installed via source and by ros.
I think it might cause some issues. Is there a solution ? @saching13