luxonis/depthai-ros #107: Bounding Box msg changed in Humble

hardesh, 2022-05-06:
Hi,

I was trying to build the repo with ROS2 Humble and found out that the bbox msg was changed again! Ref: https://github.com/ros-perception/vision_msgs/blob/ros2/msg/Pose2D.msg 

They've added a new position field in the msg. I've fixed the issue here https://github.com/Peppermint-Robots/depthai-ros 
If you want I can open up a PR

saching13, 2022-05-06:
Thanks. 
Yes. A PR would be great.

dsakshay, 2022-05-17:
@hardesh @saching13 there is also no devel support for Rolling. I observed an issue while building depthai_ros from the main branch where the build is failing because of the changes in the vision_msgs/msg/ObjectHypothesis.msg (https://github.com/ros-perception/vision_msgs/blob/ros2/msg/ObjectHypothesis.msg). This is handled in the code for galactic but not for rolling.

closed 2022-05-27 by saching13