luxonis/depthai-ros #123: oak-D S2 FPS

skyaid, 2022-06-23:
Hello 

i'm  using oak-D S2 camera and roslaunch stereo_nodelet.launch  file..

because i need to /stereo_publisher/right/image , /stereo_publisher/left/image ros topic 

but hz is not change 

I get  only 30 hz data

How to change hz?

saching13, 2022-06-23:
Please use the `stereo_inertial_node` it has the fps argument for the launch [file](https://github.com/luxonis/depthai-ros-examples/blob/main/depthai_examples/launch/stereo_inertial_node.launch)

skyaid, 2022-07-01:
@saching13 I used stereo_inertial_node.launch

The stereo depth point is posted as 10 Hz

And S2 can't be changed to "monoResolution" default=400p, is that the way it is?

saching13, 2022-08-10:
> Is sterefps max 30hz?

No. Should be able to go to 60 without RGB alignment.

> The stereo depth point is posted as 10 Hz

Thats due to host processing power and high computation needed for that I guess. 

>  And S2 can't be changed to "monoResolution" default=400p, is that the way it is?

you can.

closed 2022-08-19 by youngjae41