luxonis/depthai-ros #125: Compile Error - multiple definition of dai::ros::getFrameTime

msardonini, 2022-07-13:
I believe the function `dai::ros::getFrameTime` in `depthaiUtility.hpp` needs to be inlined. I'm getting this compile error when including this header file from more than 1 source file at a time.

```
/usr/bin/ld: CMakeFiles/oakd_s2_src.dir/src/oakd.cpp.o: in function `dai::ros::getFrameTime(rclcpp::Time, std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >)':
oakd.cpp:(.text+0x180): multiple definition of `dai::ros::getFrameTime(rclcpp::Time, std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >)'; CMakeFiles/oakd_s2_src.dir/src/oakd_s2_src.cpp.o:oakd_s2_src.cpp:(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/oakd_s2_src.dir/build.make:270: oakd_s2_src] Error 1
make[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/oakd_s2_src.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

For now, I am able to get around this using PIMPL, but it's a bit of a hassle

saching13, 2022-07-13:
@msardonini  this happens when you include `depthaiUtility.hpp` in a header file which is included in another ?

msardonini, 2022-07-13:
@saching13 not quite, this happens when `depthaiUtility.hpp` is included in two different .cpp files that are both independently compiled.

For example, If I have:

depthai_ros_publisher.hpp  <--- includes `depthaiUtility.hpp`
depthai_ros_publisher.cpp  <--- includes `depthai_ros_publisher.hpp`
main.cpp                             <--- includes `depthai_ros_publisher.hpp`

and I compile `main.cpp` and `depthai_ros_publisher.cpp`, this error will occur.

saching13, 2022-07-14:
Got it. WIll check and get back. Thanks for reporting.

saching13, 2022-08-12:
[Fixed](https://github.com/luxonis/depthai-ros/pull/129). Thanks

closed 2022-08-12 by saching13