luxonis/depthai-ros #126: Stereo nodelet fails on launch

dolf321, 2022-07-31:
[image]

Luxonis-Brandon, 2022-08-01:
Does this run for some time or die on launch?

Can you provide more details on the hardware configuration/etc.?

saching13, 2022-08-02:
@dolf321 Can you provide more logs, errors. 
And also please try `sterei_inertial_node` it has more options.

hypadr1v3, 2022-08-12:
I am having the same issue:
Running my OAK-D on a Nvidia Jetson AGX Xavier. I followed all the instructions and then ran `roslaunch depthai_examples stereo_inertial_node.launch` and got:

```
[ERROR] [1660303764.484463886]: Failed to load nodelet [/depth_image_convertion_nodelet] of type [depth_image_proc/convert_metric] even after refreshing the cache: According to the loaded plugin descriptions the class depth_image_proc/convert_metric with base class type nodelet::Nodelet does not exist. Declared types are  depthai_examples/StereoNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid
[ERROR] [1660303764.484744352]: The error before refreshing the cache was: According to the loaded plugin descriptions the class depth_image_proc/convert_metric with base class type nodelet::Nodelet does not exist. Declared types are  depthai_examples/StereoNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid
[FATAL] [1660303764.485791910]: Failed to load nodelet '/depth_image_convertion_nodelet` of type `depth_image_proc/convert_metric` to manager `nodelet_manager'
[ERROR] [1660303764.533186330]: Failed to load nodelet [/depth_image_to_rgb_pointcloud] of type [depth_image_proc/point_cloud_xyzrgb] even after refreshing the cache: According to the loaded plugin descriptions the class depth_image_proc/point_cloud_xyzrgb with base class type nodelet::Nodelet does not exist. Declared types are  depthai_examples/StereoNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid
[ERROR] [1660303764.533654617]: The error before refreshing the cache was: According to the loaded plugin descriptions the class depth_image_proc/point_cloud_xyzrgb with base class type nodelet::Nodelet does not exist. Declared types are  depthai_examples/StereoNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid
[FATAL] [1660303764.534267650]: Failed to load nodelet '/depth_image_to_rgb_pointcloud` of type `depth_image_proc/point_cloud_xyzrgb` to manager `nodelet_manager'
nnPath ,, /home/user/code/ros-depthai-example/src/luxonis/depthai-ros-examples/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
[depth_image_convertion_nodelet-5] process has died [pid 13225, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load depth_image_proc/convert_metric nodelet_manager image_raw:=/stereo_inertial_publisher/stereo/depth image:=/stereo_inertial_publisher/stereo/image __name:=depth_image_convertion_nodelet __log:=/home/nazar/.ros/log/02f268b2-1a32-11ed-ae13-0242b605c2fe/depth_image_convertion_nodelet-5.log].
[... cut ...]
================================================================================REQUIRED process [stereo_inertial_publisher-3] has died!
process has died [pid 13219, exit code -11, cmd /home/user/code/ros-depthai-example/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/user/.ros/log/02f268b2-1a32-11ed-ae13-0242b605c2fe/stereo_inertial_publisher-3.log].
log file: /home/user/.ros/log/02f268b2-1a32-11ed-ae13-0242b605c2fe/stereo_inertial_publisher-3*.log
Initiating shutdown!
================================================================================
```

saching13, 2022-08-12:
Did you install the dependencies using rosdep as in step 5 ? Looks like nodelet packages and IMU visualization tool and alla re missing in your installation. 
`rosdep install --from-paths src --ignore-src -r -y`

hypadr1v3, 2022-08-13:
Rosdep says `#All required rosdeps installed successfully` and that's it. I went ahead and just installed those packages manually instead. After all of that it runs for like 3-4 seconds but crashes again without any errors that I can see. The logs it's pointing to doesn't exist either. Only erroneous thing I can see is the `process has died`

saching13, 2022-08-13:
Check your opencv c++ version. If it is 3.2 or less update it to the newest.

hypadr1v3, 2022-08-15:
My opencv version is 4.5.4

saching13, 2022-08-16:
@daxoft  can you test this ?

saching13, 2022-08-16:
```
Failed to load nodelet [/depth_image_convertion_nodelet] of type [depth_image_proc/convert_metric] even after refreshing the cache: According to the loaded plugin descriptions the class depth_image_proc/convert_metric with base class type nodelet::Nodelet does not exist. Declared types are  depthai_examples/StereoNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid
[ERROR] [1660303764.484744352]: The error before refreshing the cache was: According to the loaded plugin descriptions the class depth_image_proc/convert_metric with base class type nodelet::Nodelet does not exist. Declared types are  depthai_examples/StereoNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid
[FATAL] [1660303764.485791910]: Failed to load nodelet '/depth_image_convertion_nodelet` of type `depth_image_proc/convert_metric` to manager `nodelet_manager'
[ERROR] [1660303764.533186330]: Failed to load nodelet [/depth_image_to_rgb_pointcloud] of type [depth_image_proc/point_cloud_xyzrgb] even after refreshing the cache: According to the loaded plugin descriptions the class depth_image_proc/point_cloud_xyzrgb with base class type nodelet::Nodelet does not exist. Declared types are  depthai_examples/StereoNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid
[ERROR] [1660303764.533654617]: The error before refreshing the cache was: According to the loaded plugin descriptions the class depth_image_proc/point_cloud_xyzrgb with base class type nodelet::Nodelet does not exist. Declared types are  depthai_examples/StereoNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid
[FATAL] [1660303764.534267650]: Failed to load nodelet '/depth_image_to_rgb_pointcloud` of type `depth_image_proc/point_cloud_xyzrgb` to manager `nodelet_manager'
```

@hypadr1v3  not sure why but your system is not able to find the nodelets from different packages of ROS which are in use. Can you check why your setup is not able to find them ?

here is the package link http://wiki.ros.org/depth_image_proc

daxoft, 2022-08-17:
I did a fresh install on my Nvidia Xavier AGX with Ubuntu20, and it worked both with packages installed from ROS Binaries and compiled locally, for ROS Noetic, Galactic and Foxy. 

@hypadr1v3 what OS version and ROS version you have installed on your Nvidia Jetson? Did you build from sources or installed from ROS Binaries?

hypadr1v3, 2022-08-17:
I'm also on 20 with ROS noetic. These errors came when I compiled from source. Now I went and tried the binaries and they seem to work perfectly...

Serafadam, 2022-11-07:
Seems to be solved, so closing, if you think this should remain open, please tag me in a comment here.

closed 2022-11-07 by Serafadam