luxonis/depthai-ros #128: Unable to create 3D map with OAK-D

Adarsh-S-Bharadwaj, 2022-08-10:
I am using ROS1 Ubuntu 18.04 ROS Melodic. Unable to create a 3D map with OAK D when launched with Rtabmap with all the arguments as mentioned in Rtabmap stereo handheld mapping, I am using MPU9150 imu. I am also having a transform link between oak_d-frame and base_footprint. I am using stereo_inertial_node.launch file of depthai-ros package for launching OAK D camera. Any help or solution regarding this problem is appreciated.

[image]

[image]

Thank you

saching13, 2022-08-10:
Let me know if [this](https://github.com/introlab/rtabmap/issues/742#issuecomment-1059600320) helps.

Adarsh-S-Bharadwaj, 2022-08-11:
Thank you for your response @saching13 but that is not working for me. I am using the same updated bringup and rtabmap launch for OAK-D stereo mapping which is mentioned on this website. http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping
I tried two OAK-D cameras separately, one with IMU and one without IMU but still, it gives the same error in both cameras.

saching13, 2022-08-11:
Thanks. Will check on our side and get back to you.

saching13, 2022-08-25:
Sorry was not able to get it to working. looks like there are internal issues on Rtabmap. Need more time to debug. 
I would be addressing adding other SLAM and VIO examples first and get back to this.

Adarsh-S-Bharadwaj, 2022-08-26:
Sure. Thank you.

matlabbe, 2022-09-26:
Tried example from http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping with my OAK-D, and worked with latest `rtabmap_ros` version and also the `ros-noetic-rtabmap-ros` binaries on noetic. I tested with `ros-noetic-depthai-ros` binaries.

If it still don't work, things to check, after doing:
```
roslaunch depthai_examples stereo_inertial_node.launch depth_aligned:=false

rosrun imu_filter_madgwick imu_filter_node    \
   imu/data_raw:=/stereo_inertial_publisher/imu    \
   imu/data:=/stereo_inertial_publisher/imu/data     \
   _use_mag:=false    \
   _publish_tf:=false
```
do:
```
rostopic hz /stereo_inertial_publisher/left/image_rect \
   /stereo_inertial_publisher/right/image_rect \
   /stereo_inertial_publisher/left/camera_info \
   /stereo_inertial_publisher/right/camera_info \
   /stereo_inertial_publisher/imu/data

subscribed to [/stereo_inertial_publisher/left/image_rect]
subscribed to [/stereo_inertial_publisher/right/image_rect]
subscribed to [/stereo_inertial_publisher/left/camera_info]
subscribed to [/stereo_inertial_publisher/right/camera_info]
subscribed to [/stereo_inertial_publisher/imu/data]
                   topic                        rate   min_delta   max_delta   std_dev    window
================================================================================================
/stereo_inertial_publisher/left/image_rect     15.02   0.06438     0.06924     0.001187   14    
/stereo_inertial_publisher/right/image_rect    14.95   0.06405     0.0708      0.002087   14    
/stereo_inertial_publisher/left/camera_info    15.01   0.06357     0.06982     0.001641   15    
/stereo_inertial_publisher/right/camera_info   14.98   0.06309     0.0711      0.002237   15    
/stereo_inertial_publisher/imu/data            389.0   3.338e-06   0.0299      0.005644   14
```
Make sure topics are published! and a with a decent rate... though not sure why it is 15 Hz on my computer right now, but it still work.

saching13, 2022-09-28:
Thanks @matlabbe

ParsaKhaledi, 2022-10-04:
After the latest update published 3 days ago I still have ["oak_imu_frame" doesn't exist ] problem.
I have OAK-D-Pro and used launch file configuration mentioned above with ros2. Also, there is no data on /tf topic.
@matlabbe
`
ros2 run imu_filter_madgwick imu_filter_madgwick_node \
   --ros-args \
   -p use_mag:=false \
   -p publish_tf:=false \
   -p world_frame:="enu" \
   -r /imu/data_raw:=/imu \
   -r /imu/data:=/rtabmap/imu

ros2 launch rtabmap_ros  \
    args:="--delete_db_on_start" \
    stereo:=true \
    left_image_topic:=/left/image_rect \
    right_image_topic:=/right/image_rect \
    left_camera_info_topic:=/left/camera_info \
    right_camera_info_topic:=/right/camera_info \
    imu_topic:=/rtabmap/imu \
    frame_id:=oak-d_frame \
    approx_sync:=true \
    approx_sync_max_interval:=0.001 \
    wait_imu_to_init:=true \
    qos:=1

`
---------------------------------------------------------------------
Terminal Output:

`~/ros2_ws$ ros2 launch rtabmap_ros rtabmap.launch.py     args:="--delete_db_on_start"     stereo:=true     left_image_topic:=/left/image_rect     right_image_topic:=/right/image_rect     left_camera_info_topic:=/left/camera_info     right_camera_info_topic:=/right/camera_info     imu_topic:=/rtabmap/imu     frame_id:=oak-d-base-frame     approx_sync:=true     approx_sync_max_interval:=0.001     wait_imu_to_init:=true     qos:=1
[INFO] [launch]: All log files can be found below /home/droneai/.ros/log/2022-10-04-13-58-13-626687-asus-13024
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [stereo_odometry-1]: process started with pid [13026]
[INFO] [rtabmap-2]: process started with pid [13028]
[INFO] [rtabmapviz-3]: process started with pid [13030]
[stereo_odometry-1] [INFO] [1664879293.885076717] [rtabmap.stereo_odometry]: Odometry: frame_id               = oak-d-base-frame
[stereo_odometry-1] [INFO] [1664879293.885194067] [rtabmap.stereo_odometry]: Odometry: odom_frame_id          = odom
[stereo_odometry-1] [INFO] [1664879293.885201672] [rtabmap.stereo_odometry]: Odometry: publish_tf             = true
[stereo_odometry-1] [INFO] [1664879293.885207106] [rtabmap.stereo_odometry]: Odometry: wait_for_transform     = 0.200000
[stereo_odometry-1] [INFO] [1664879293.885235770] [rtabmap.stereo_odometry]: Odometry: initial_pose           = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[stereo_odometry-1] [INFO] [1664879293.885244131] [rtabmap.stereo_odometry]: Odometry: ground_truth_frame_id  = 
[stereo_odometry-1] [INFO] [1664879293.885249428] [rtabmap.stereo_odometry]: Odometry: ground_truth_base_frame_id = 
[stereo_odometry-1] [INFO] [1664879293.885254742] [rtabmap.stereo_odometry]: Odometry: config_path            = 
[stereo_odometry-1] [INFO] [1664879293.885259748] [rtabmap.stereo_odometry]: Odometry: publish_null_when_lost = true
[stereo_odometry-1] [INFO] [1664879293.885265408] [rtabmap.stereo_odometry]: Odometry: guess_frame_id         = 
[stereo_odometry-1] [INFO] [1664879293.885270662] [rtabmap.stereo_odometry]: Odometry: guess_min_translation  = 0.000000
[stereo_odometry-1] [INFO] [1664879293.885276711] [rtabmap.stereo_odometry]: Odometry: guess_min_rotation     = 0.000000
[stereo_odometry-1] [INFO] [1664879293.885282624] [rtabmap.stereo_odometry]: Odometry: guess_min_time         = 0.000000
[stereo_odometry-1] [INFO] [1664879293.885288358] [rtabmap.stereo_odometry]: Odometry: expected_update_rate   = 0.000000 Hz
[stereo_odometry-1] [INFO] [1664879293.885293994] [rtabmap.stereo_odometry]: Odometry: max_update_rate        = 0.000000 Hz
[stereo_odometry-1] [INFO] [1664879293.885299670] [rtabmap.stereo_odometry]: Odometry: wait_imu_to_init       = true
[stereo_odometry-1] [INFO] [1664879293.885362525] [rtabmap.stereo_odometry]: Odometry: stereoParams_=1 visParams_=1 icpParams_=0
[stereo_odometry-1] [INFO] [1664879293.937196272] [rtabmap.stereo_odometry]: odometry: Subscribing to IMU topic /rtabmap/imu
[stereo_odometry-1] [INFO] [1664879293.937228380] [rtabmap.stereo_odometry]: odometry: qos_imu = 1
[stereo_odometry-1] [INFO] [1664879293.937541830] [rtabmap.stereo_odometry]: StereoOdometry: approx_sync = true
[stereo_odometry-1] [INFO] [1664879293.937552094] [rtabmap.stereo_odometry]: StereoOdometry: approx_sync_max_interval = 0.001000
[stereo_odometry-1] [INFO] [1664879293.937561837] [rtabmap.stereo_odometry]: StereoOdometry: queue_size  = 10
[stereo_odometry-1] [INFO] [1664879293.937567412] [rtabmap.stereo_odometry]: RGBDOdometry: qos            = 1
[stereo_odometry-1] [INFO] [1664879293.937572840] [rtabmap.stereo_odometry]: RGBDOdometry: qos_camera_info = 1
[stereo_odometry-1] [INFO] [1664879293.937578047] [rtabmap.stereo_odometry]: StereoOdometry: subscribe_rgbd = false
[stereo_odometry-1] [INFO] [1664879293.937583320] [rtabmap.stereo_odometry]: StereoOdometry: keep_color     = false
[rtabmap-2] [INFO] [1664879293.941477349] [rtabmap.rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[rtabmap-2] [INFO] [1664879293.941625047] [rtabmap.rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[rtabmap-2] [INFO] [1664879293.941635578] [rtabmap.rtabmap]: rtabmap(maps): map_cleanup                = true
[rtabmap-2] [INFO] [1664879293.941641730] [rtabmap.rtabmap]: rtabmap(maps): map_always_update          = false
[rtabmap-2] [INFO] [1664879293.941646570] [rtabmap.rtabmap]: rtabmap(maps): map_empty_ray_tracing      = true
[... cut ...]
[stereo_odometry-1] Warning: Invalid frame ID "oak_imu_frame" passed to canTransform argument source_frame - frame does not exist
[stereo_odometry-1]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[stereo_odometry-1] Warning: Invalid frame ID "oak_imu_frame" passed to canTransform argument source_frame - frame does not exist
[stereo_odometry-1]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rtabmap-2] [INFO] [1664879294.144278104] [rtabmap.rtabmap]: rtabmap: Database version = "0.20.21".`

matlabbe, 2022-10-04:
I guess you are referring to this example: https://github.com/introlab/rtabmap/issues/742#issuecomment-1259965385

Got an issue (not the one you have) explained in the following post but it is now fixed in latest master branch rtabmap_ros 0.20.20 (will soon be released as binaries). Overall, I didn't see your issue. Can you tell us the exact version of `depthai-ros` used?

matlabbe, 2022-10-05:
You could launch with `wait_imu_to_init:=true` to ignore imu. I don't have an Oak-D Pro, but if there is an imu topic, maybe the corresponding TF is not correctly published by depthai node. What is the TF tree?
```
ros2 run tf2_tools view_frames
```

Here is mine with normal Oak-D:
[image]

ParsaKhaledi, 2022-10-06:
Actually, I believe this is the problem. I don't have oak_imu_frame. I also think this is why I cannot run Rtabmap. Also, I asked a friend to test OAK-D-S2; the result was the same as the picture below.
[image]

And thanks for your reply.

ParsaKhaledi, 2022-10-06:
Update:
I use camera_model:=OAK-D instead of camera_model:=OAK-D-PRO and then I have oak_imu_frame. 
'''
ros2 launch depthai_examples stereo_inertial_node.launch.py camera_model:=OAK-D
'''
[image]

an99990, 2022-10-10:
I am still having some issues in ros1 with the oak-d camera. This happened after i decided to build rtabmap with DEPTHAI_CORE=ON, i also have two opencv version 4.0 and 3.4.9. I had to install 4.0 becaue depth core ai needed it. is the issue related to the opencv versions ?
ros : melodic
please help @matlabbe 

```
roslaunch rtabmap_ros rtabmap.launch args:="--delete_db_on_start" \
> rgb_topic:=/stereo_inertial_publisher/color/image \
> camera_info_topic:=/stereo_inertial_publisher/color/camera_info \
> depth_topic:=/stereo_inertial_publisher/stereo/depth \
> imu_topic:=/stereo_inertial_publisher/imu \
> frame_id:=oak-d_frame \
> approx_sync:=true \
> approx_sync_max_interval:=0.001 \
> wait_imu_to_init:=true

... logging to /home/ubuntu/.ros/log/d8584b62-48e9-11ed-a7ff-bf1d1f317710/roslaunch-ubuntu-3271.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.154:33355/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmapviz/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/approx_sync_max_interval: 0.001
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/expected_update_rate: 0.0
 * /rtabmap/rgbd_odometry/frame_id: oak-d_frame
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/keep_color: False
 * /rtabmap/rgbd_odometry/max_update_rate: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rgbd_odometry/wait_imu_to_init: True
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: oak-d_frame
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
[... 238 lines cut ...]
imu_topic:=/stereo_inertial_publisher/imu \
frame_id:=oak-d_frame \
approx_sync:=true \
approx_sync_max_interval:=0.001 \
wait_imu_to_init:=true
[ WARN] [1665440584.484688980]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /stereo_inertial_publisher/color/image \
   /stereo_inertial_publisher/stereo/depth \
   /stereo_inertial_publisher/color/camera_info \
   /rtabmap/odom_info
[ WARN] [1665440585.985665026]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmapviz subscribed to (approx sync):
   /rtabmap/odom \
   /stereo_inertial_publisher/color/image \
   /stereo_inertial_publisher/stereo/depth \
   /stereo_inertial_publisher/color/camera_info \
   /rtabmap/odom_info
^C[rtabmap/rtabmapviz-3] killing on exit
[ INFO] [1665440587.865340018]: rtabmapviz: ctrl-c catched! Exiting Qt app...
[rtabmap/rtabmap-2] killing on exit
rtabmap: Saving database/long-term memory... (located at /home/ubuntu/.ros/rtabmap.db)
rtabmap: Saving database/long-term memory...done! (located at /home/ubuntu/.ros/rtabmap.db, 0 MB)
```

an99990, 2022-10-12:
i rebuilt rtbamap without depthai_core and still got the error mentioned above. Does rtabmap support opencv version higher than 3.2.I did get warnings of potential conflicts of version @matlabbe 
thank you for your help

matlabbe, 2022-10-23:
@an99990 If you mix up many opencv versions on your computer, it is likely that it may cause some crashes in libraries/apps using opencv. To use latest OpenCV version, you will have to unsintall opencv binaries from your computer. This will uninstall `cv_bridge` package and all packages depending on it, so you would have to rebuild all those packages from source so that they all link on that opencv version. To use opencv modules that are not installed with the binaries, a workaround is to rebuild the same opencv version than the binaries on your platform (see comment [here](https://github.com/introlab/rtabmap_ros#build-from-source) about rebuilding opencv).