luxonis/depthai-ros #132: oak-d pro DotProjector and floodLightmA is not running

youngjae41, 2022-08-19:
I have oak-D-Pro and oak-D-PRO POE version 

I used stereo_inertial_node.launch 

   <arg name="enableDotProjector"    default="true"/>
    <arg name="enableFloodLight"      default="true"/>
    <arg name="dotProjectormA"        default="800.0"/> or default="200.0"
    <arg name="floodLightmA"          default="200.0"/>

DotProjector and floodLightmA is not running

In depthai_demo.py it works fine

saching13, 2022-08-25:
it is running increase the values. Make it `1000` for example

youngjae41, 2022-08-25:
@saching13 thanks

How is the maximum ?

saching13, 2022-08-28:
> @saching13 thanks
> 
> 
> 
> How is the maximum ?

I didn't get you.

youngjae41, 2022-08-29:
@saching13 setting it to 1000 doesn't work

saching13, 2022-08-29:
Ok. let me check and get back

youngjae41, 2022-08-29:
Thanks @saching13 

and there is a difference of about 7 frames in POE type and USB type FPS

saching13, 2022-09-06:
@youngjae41  check [this](https://github.com/luxonis/depthai-ros/issues/85) please

saching13, 2022-09-06:
and can you record a video of your floodLight and share it with us ?

saching13, 2022-09-06:
### With FloodLight

[image]

### Without FloodLight
[image]

saching13, 2022-09-06:
### With Floodlight and DotProjector. 

[image]

saching13, 2022-09-06:
Sorry for the issue. We changed the boardNames internally in the new batches of manufacturing. And this new check was not added here. I fixed and updated the code. it should work now. @youngjae41

closed 2022-09-06 by saching13

youngjae41, 2022-09-08:
@saching13 Thanks 

work now!!