luxonis/depthai-ros #137: I have issue when running ros2 launch depthai_examples mobile_publisher.launch.py camera_model:=OAK-D

Kazimbalti, 2022-09-07:
I recieved the following error, 
Please help me to resolve this issue.
Thank you

When I am running 
ros2 launch depthai_examples mobile_publisher.launch.py camera_model:=OAK-D

I have found the following issue,
I am running on Turtlebot4 which already have OAK-D-Pro camera.

[INFO] [launch]: All log files can be found below /home/asl/.ros/log/2022-09-07-16-07-35-324006-asl-ThinkPad-P15v-Gen-2i-38811
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mobilenet_node-1]: process started with pid [38814]
[INFO] [robot_state_publisher-2]: process started with pid [38816]
[robot_state_publisher-2] Link oak-d_frame had 5 children
[robot_state_publisher-2] Link oak_imu_frame had 0 children
[robot_state_publisher-2] Link oak_left_camera_frame had 1 children
[robot_state_publisher-2] Link oak_left_camera_optical_frame had 0 children
[robot_state_publisher-2] Link oak_model_origin had 0 children
[robot_state_publisher-2] Link oak_rgb_camera_frame had 1 children
[robot_state_publisher-2] Link oak_rgb_camera_optical_frame had 0 children
[robot_state_publisher-2] Link oak_right_camera_frame had 1 children
[robot_state_publisher-2] Link oak_right_camera_optical_frame had 0 children
[robot_state_publisher-2] [INFO] [1662534455.487157898] [oak_state_publisher]: got segment oak-d-base-frame
[robot_state_publisher-2] [INFO] [1662534455.487199367] [oak_state_publisher]: got segment oak-d_frame
[robot_state_publisher-2] [INFO] [1662534455.487204565] [oak_state_publisher]: got segment oak_imu_frame
[robot_state_publisher-2] [INFO] [1662534455.487207841] [oak_state_publisher]: got segment oak_left_camera_frame
[robot_state_publisher-2] [INFO] [1662534455.487211131] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[robot_state_publisher-2] [INFO] [1662534455.487214129] [oak_state_publisher]: got segment oak_model_origin
[robot_state_publisher-2] [INFO] [1662534455.487217267] [oak_state_publisher]: got segment oak_rgb_camera_frame
[robot_state_publisher-2] [INFO] [1662534455.487220341] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[robot_state_publisher-2] [INFO] [1662534455.487223334] [oak_state_publisher]: got segment oak_right_camera_frame
[robot_state_publisher-2] [INFO] [1662534455.487226426] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[mobilenet_node-1] terminate called after throwing an instance of 'std::runtime_error'
[mobilenet_node-1]   what():  No available devices
[ERROR] [mobilenet_node-1]: process has died [pid 38814, exit code -6, cmd '/home/asl/dai_ws/install/depthai_examples/lib/depthai_examples/mobilenet_node --ros-args --params-file /tmp/launch_params_5z7p28lu --params-file /tmp/launch_params_olkejha5 --params-file /tmp/launch_params_xdnqit6b --params-file /tmp/launch_params_gpexm_ko --params-file /tmp/launch_params_3paxzld4'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-2] [INFO] [1662534580.784098941] [rclcpp]: signal_handler(signal_value=2)
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 38816]

saching13, 2022-10-10:
your device is not detected.
[Here](https://docs.luxonis.com/en/latest/pages/troubleshooting/) is the troubleshooting steps,

Serafadam, 2022-11-09:
Closing due to inactivity, if you think this should remain open, please tag me in a comment here.

closed 2022-11-09 by Serafadam